Bodmer/TFT_eSPI #169: TFT_eSPI' has no member named 'getTouch'

tgirard, 2018-07-07:
Hello, First, thanks for all this, it's a wonderful library.
I am using a NodeMCU 0.9 controller
The screen is a standard 2.8TFT,  I9341 STM32 based Touch enabled screen (Using an HR2046 Chip)

I did a clean install of the library, changed DC and CS for my setup and ran a couple of graphics test to be sure that the library would compile, send to my controller and display, which it does.

When I try to run a touch screen demo I get the "TFT_eSPI' has no member named 'getTouch'" error
I see it in the Keywords so I know it's there.

Is there something in the setup.h file that I'm missing? I didn't see anything specific
Any help would be greatly appreciated
Thanks
-Tim

Bodmer, 2018-07-07:
If you define a pin for the touch chip select then it should be OK:

https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L97

tgirard, 2018-07-07:
That was exactly it
#define TOUCH_CS PIN_D0     // Chip select pin (T_CS) of touch screen

Thank you so much for getting back to me in time for me to get my project done this weekend :)

closed 2018-07-07 by Bodmer

MohammedKarro, 2019-10-08:
> If you define a pin for the touch chip select then it should be OK:
> 
> https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L97

I am using the  ILI9481_DRIVER
the Pins are as following 
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

I dont't get it, isn't the " #define TFT_CS   33 " resonsible for the Touch?

KedarAMare, 2020-12-03:
For the touch screen to work with ESP32 you must un-comment the line below  in the “User_Setup.h” file.

#define TOUCH_CS 21 // Chip select pin (T_CS) of touch screen

drphil3d, 2021-01-26:
I have this issue on custom hardware I've developed. I used GPIO 0 for CS and I'm getting this same error.

Bodmer, 2021-01-26:
Run the diagnostic sketch Read_User_Setup and check the compiler is picking up your settings by looking at the Serial Monitor output. Note that touch only works with SPI displays and XPT2046 touch controllers. In other cases an alternative touch library can be used.

drphil3d, 2021-01-27:
Touch Screen seems to work fine. The problem must be elsewhere. I just don't know my program isn't reading the config file.
[code]
TFT_eSPI ver = 2.3.56
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 0

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
[/code]

Bodmer, 2021-01-27:
I see you have an ILI9488 display. Make sure you have not connected the MISO signal only to the touch controller, not the TFT SDO pin (see[ ILI9488 part of discussion here](https://github.com/Bodmer/TFT_eSPI/discussions/898)).

Run the diagnostic sketch  Test_Touch_Controller. This does not output to the TFT screen (it will reman blank) the the raw touch output will be sent to the serial Monitor Window. Touch the screen and see if the values change significantly. Z is a measure of the touch pressure on the screen. Post a smal segment of the output here.

drphil3d, 2021-01-27:
Oh geez, I didn't even catch that. I'm actually using an ST7796. I got my config files mixed up.
I had this display custom manufactured, it has a 1000nits backlight + OCA bonded RTP.  I'm in the process of importing the project into PlatformIO from Arduino. It will be much easier to use build flags instead of config files.

Starting...
x: 0     y: 4095     z: 160 
x: 0     y: 4095     z: 152 
x: 0     y: 4095     z: 151 
x: 1339     y: 2658     z: 1667 
x: 1956     y: 2681     z: 1838 
x: 2601     y: 2170     z: 2246 
x: 2087     y: 1671     z: 2160 
x: 1441     y: 2359     z: 1850 

[code]
TFT_eSPI ver = 2.3.56
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7796
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 0

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
[/code]

drphil3d, 2021-01-27:
Touch Calibration works now, so I suspect that error is coming from something in my code.

Bodmer, 2021-01-27:
OK,so all is good now. Hope your project goes well.

BabkoAndrey, 2022-01-06:
Hello! Help me pls. If i have add #define TOUCH_CS 21 , i have Compilation error for ESP32 Dev Module board. What can i do ? Pls help! Thank you

Bodmer, 2022-01-06:
Run the diagnostic sketch Read_User_Setup and check the compiler is picking up your settings.

BabkoAndrey, 2022-01-06:
On the read_user_setup I have add this code :
--https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Test%20and%20diagnostics/Read_User_Setup/Read_User_Setup.ino
But dont work
Can not install on esp32
Отправлено из myMail для Android

BabkoAndrey, 2022-01-06:
Compilation error for ESP32(((

Bodmer, 2022-01-06:
Have you successfully run the Basics>Blink example to flash an LED?

BabkoAndrey, 2022-01-06:
I have write scatch for termocouple and screen shows temperature very well! But dont work skatch with touch((

Bodmer, 2022-01-06:
So the sketch and display are running unless a touch pin is defined, then you get a compile error?  Post the compile error.

BabkoAndrey, 2022-01-06:
/Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master/Extensions/Touch.cpp:22:34: error: 'spi' was not declared in this scope
     if (locked) {locked = false; spi.beginTransaction(SPISettings(SPI_TOUCH_FREQUENCY, MSBFIRST, SPI_MODE0));}
                                  ^
/Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master/Extensions/Touch.cpp: In member function 'void TFT_eSPI::end_touch_read_write()':
/Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master/Extensions/Touch.cpp:37:54: error: 'spi' was not declared in this scope
     if(!inTransaction) {if (!locked) {locked = true; spi.endTransaction();}}
                                                      ^
/Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master/Extensions/Touch.cpp: In member function 'uint8_t TFT_eSPI::getTouchRaw(uint16_t*, uint16_t*)':
/Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master/Extensions/Touch.cpp:61:3: error: 'spi' was not declared in this scope
   spi.transfer(0xd0);                    // Start new YP conversion
   ^
/Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master/Extensions/Touch.cpp: In member function 'uint16_t TFT_eSPI::getTouchRawZ()':
/Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master/Extensions/Touch.cpp:104:3: error: 'spi' was not declared in this scope
   spi.transfer(0xb0);               // Start new Z1 conversion
   ^
Используем библиотеку FS версии 1.0 из папки: /Users/andrey/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/FS 
Используем библиотеку SPI версии 1.0 из папки: /Users/andrey/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPI 
Используем библиотеку TFT_eSPI-master версии 2.4.22 из папки: /Users/andrey/Documents/Arduino/libraries/TFT_eSPI-master 
Используем библиотеку SPIFFS версии 1.0 из папки: /Users/andrey/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPIFFS 
exit status 1
Ошибка компиляции для платы ESP32 Dev Module.

Bodmer, 2022-01-06:
Are you using a parallel display?

If so then the library does not support touch screens when a parallel display is used.

BabkoAndrey, 2022-01-06:
Oh, maybe! So if parallel display how can i used touch, or better buy not parallel?

Bodmer, 2022-01-06:
You can use a touch library, the choice depends on the touch interface chip fitted. Some parallel displays, typically in UNO board form have a resistance only output, others have chips fitted.

BabkoAndrey, 2022-01-06:
Ok! Thank you for answer! And maybe you can send me link for lib for parallel ILI9341

Bodmer, 2022-01-06:
That is the display driver chip, not the touch driver chip.

Post a link to the display sellers page.

BabkoAndrey, 2022-01-06:
https://uamper.com/index.php?route=product/product&path=182&product_id=23&gclid=Cj0KCQiAw9qOBhC-ARIsAG-rdn7rgW_T0QmefzuaruuFJxZVVMSVqPiP78-YBqaa20a0CMSiZbEzUC8aAgbNEALw_wcB

miky-boy, 2022-03-10:
Hi,

I'm also interested on this topic. I bought the "Arduino 3.5 inch TFT LCD Display Touch Screen UNO R3 Board Plug and Play" from this URL https://www.ebay.com/itm/373700584024?mkevt=1&mkpid=0&emsid=e11051.m43.l1123&mkcid=7&ch=osgood&euid=be5f00f68f2c4e0fbf6a9c809ebd6ed9&bu=44288059471&osub=-1%7E1&crd=20220309053229&segname=11051&sojTags=ch%3Dch%2Cbu%3Dbu%2Cosub%3Dosub%2Ccrd%3Dcrd%2Csegname%3Dsegname%2Cchnl%3Dmkcid 

I was using directly this TFT screen on a Wemos ESP32 board and with the 3 wires hardware mod i got sucess and the image start appearing on this screen. At the beguinning mirrored and after a lot of searches i found the correct drive for my TFT is the below one:
ID = 0x5310
reg(0x00D4) 01 01 53 10	Novatek ID 

To make your library work with my TFT i need to force the usage of #define ST7735_DRIVER but changing the dimenstions to 480*320 because with any other i always got the image mirrored :( I don't know if you know the reason for this, but according what i search seems to be the need of having the "INVERT_SS" command of init.

My huge problem now is related with the touch functionality. I tested with the "MCUFRIEND_kbv" and made the callibration but i got strange measures and i can't have the touch "synchronized" with the image on screen. Exists always some difference.

Testing with your libraries the touch funcionality i got first the error "TFT_eSPI' has no member named 'getTouch'". I have uncomented the line #define TOUCH_CS 21, but i'm not sure if the port number is correct. Where I can find this port? And my TFT use paralell ports, so i got after the same error as mentioned above about the spi ports.

I think i'm on a dead end. Do you have any idea or could guide me to any possible solution please?

Thanks in advice and Best Regards,
Miky

Bodmer, 2022-03-10:
This library does not support touch screens on a parallel interface TFT. That is why the error message is generated.

As noted in the ReadMe, if the display board is fitted with a resistance based touch screen then this can be used by performing the modifications described in the ReadMe and the also the additional modification described in the following fork of the Adafruit library:
https://github.com/s60sc/Adafruit_TouchScreen

Bodmer, 2022-03-10:
The web link indiicates you board uses an ILI9488 chip, so use the correct driver in the setup file.

miky-boy, 2022-03-10:
> This library does not support touch screens on a parallel interface TFT. That is why the error message is generated.
>
> As noted in the ReadMe, if the display board is fitted with a resistance based [...]

Hi my friend,
Thanks for your reply. i already followed this library you mentioned and made the hardware mods. This hardware mods made the LCD start appearing something, because before i always getting the white screen on LCD.

miky-boy, 2022-03-10:
> The web link indicates your board uses an ILI9488 chip, so use the correct driver in the setup file.

Hi my friend,
I'm not sure. If i use the #define ILI9488_DRIVER  i have image on screen but always mirrored. I tried several ways and libraries and only the ST7735_DRIVER work properly with the size changed to 480*320. 
I tested with the LCD_ID_readreg and got the results below :(
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 80 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 66 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 0C 0C	Inversion Control
reg(0x00B6) 02 02 08 08 08	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 44 44 44 10 10 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 80 80 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 01 01 53 10	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 80 80	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) F1 F1 54 A5 24 1E 44 90 02 2C 2C 32	Adjust Control 2
reg(0x00F6) 00 00 3A 00	Interface Control

mdabdillah, 2023-12-14:
For touchsccreen for SAMD21, what should I uncomment?

ramiss, 2024-01-23:
I will add that I struggled with this issue for a few hours.  I started with a parallel set and had  -D ESP32_PARALLEL=1.   When I switched to an SPI display I logically thought I could just set  -D ESP32_PARALLEL=0.  However, that apparently doesn't load the Touch.h file.  Any mention of this line will cause it to not load.