Bodmer/TFT_eSPI #170: Problem with HX8357D monitor with ESP32

zrefai, 2018-07-10:
I've tried everything, but I cant seem to get this display to work with my ESP32 in parallel:
https://learn.adafruit.com/adafruit-3-5-color-320x480-tft-touchscreen-breakout/overview

I was running the 480x320 TFT_graphicstest_one_lib and my serial monitor posted this:

```
⸮⸮⸮⸮⸮V⸮⸮K`⸮

TFT_eSPI library test!
Benchmark                Time (microseconds)
Screen fill              201960
Text                     6499
Lines                    75665
Horiz/Vert Lines         18925
Rectangles (outline)     10606
Rectangles (filled)      562857
Circles (filled)         71822
Circles (outline)        41637
Triangles (outline)      16658
Triangles (filled)       183661
Rounded rects (outline)  23089
Rounded rects (filled)   592404
Done!
```

I think the problem is that its not sensing the chip on the screen, but I could be wrong. Any suggestions? My User_Setup.h file has the HX8357D driver and #define ESP32_PARALLEL un-commented, and I've double checked my wiring to the correct pins.

Bodmer, 2018-07-10:
The library does not read the display type so relies on the setup being correct.  Run the "Read_User_Setup" sketch and look at the serial output to make sure the compiler is picking up the setup configuration expected.
 Check the power and ground connections are OK and that you have a backlight glow from the screen.

As the display is an Adafruit product it would be worth trying their library by following the tutorial. That might help verify if it is the TST_eSPI library that is causing the issue. Users of that display have confirmed the display does work with the library in SPI mode. I do not have one of those displays so am unable to do any tests myself.

zrefai, 2018-07-18:
I've tried everything you've mentioned in your previous comment, nothing works.

Bodmer, 2018-07-18:
Following the Adafruit tutorial again, with the exact hardware setup they recommend would be the next step. One possibility is that the display is damaged or has a fault.

closed 2018-07-18 by Bodmer