Bodmer/TFT_eSPI #173: Press event

abdollahpour, 2018-07-21:
The code:

     boolean pressed = tft.getTouch(&t_x, &t_y);

returns true with an interval as long as the user keeps the figure on the screen. Is there any way to disable this behavior to create a press event?

Franciscopcosta, 2018-07-23:
i am also very interested in this issue

Franciscopcosta, 2018-07-23:
@Bodmer why this lib don use the T_IRQ pin do detect if the “button” is still pressed?
thanks a lot for you great job congratulations very fast lib

Bodmer, 2018-07-25:
The library only provides simple detection capability, more advanced behaviors must be coed in your sketch.

A simple way to implement the behaviour you want is to create a "press" variable that reflects the last detected touch state, then a "press" event is when that variable was false but a touch is detected. So effectively you are making a "change of state" indicator.

abdollahpour, 2018-07-25:
Thanks, that's exactly what I'm doing now.

Bodmer, 2018-07-25:
@Franciscopcosta
Adding the touch interrupt complicates matters and tends to trigger multiple times, so the approach I have used is to detect and filter the touch events via a simpler polling method.

There are other XPT2046 touch libraries that do use interrupts, so using one of those is an option if this is a "must have" feature.

closed 2018-07-25 by Bodmer