Bodmer/TFT_eSPI #174: TFT-sSPI Touch not working with Adafruit Featherwing (w/fix)

wtilton, 2018-07-24:
I'm using Adafruit's Featherwing https://learn.adafruit.com/adafruit-2-4-tft-touch-screen-featherwing/pinouts. Took awhile to get the display working but I did with this config.

```
#define TFT_CS   0  // Chip select control pin
#define TFT_DC   15  // Data Command control pin
#define TFT_RST  2  // Reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1  // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

#define TOUCH_CS 16     // Chip select pin (T_CS) of touch screen
```

I initially got just the display working with after running the Read_User_Setup example.  The TOUCH_CS seemed to be incorrect though for this board as it was I think telling me to set it to D2 (Pin 4). I went through Adafruit's sample and they have a touchscreen controller library that takes in a STMPE_CS value of 16. I changed the TOUCH_CS to 16 and everything works. Adafruit never mentions the TOUCH_CS for their stuff, so I really wasn't sure what to use.

I'm not sure if how you would want to handle this with either a pull request or what but this took me a few days of learning/playing with it to get it to work so I wanted to post it so that it will save others some time.

I haven't used the library yet, but it looks very promising. Thanks for all the hard work.

wtilton, 2018-07-24:
I'm also not sure if the D pins are the same as the GPIO on this board. I found a user submitted definitions of:

```
static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10  = 1;
```

wtilton, 2018-07-24:
I actually don't think this is still working even though values are changing in Test_Touch_Controller. Sample output:

x: 4047     y: 4047     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 4047     y: 4047     z: 5167 
x: 8143     y: 4047     z: 5167 
x: 8143     y: 0     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 4047     y: 0     z: 2971 
x: 899     y: 0     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 4047     y: 0     z: 5167 
x: 8143     y: 0     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 899     y: 0     z: 4095 
x: 899     y: 0     z: 4095 
x: 899     y: 0     z: 2971 
x: 4047     y: 0     z: 4095 
x: 4047     y: 0     z: 5167 
x: 8143     y: 0     z: 3003 
x: 899     y: 0     z: 2971 
x: 4047     y: 0     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 4047     y: 4047     z: 4095 
x: 4047     y: 4047     z: 4095 

I have my finger slowly moving about the center. I don't think I should be getting y values of 0 ever. Maybe as a fix I can use Adafruit's Adafruit_STMPE610.h library for now. It seems to be working so far so I don't think it is my hardware.

wtilton, 2018-07-24:
On_Off_Button also renders fine, but nothing seems to happen on the screen.

Bodmer, 2018-07-25:
The TFT_eSPI library supports the XPT2046 controller.  The Featherwing display you have uses the STMPE610 controller so this is why the TFT_eSPI examles do not work.  You will need to use the Adafruit STMPE610 library and adapt the TFT_eSPI examples to use those functions, in this case do not declare the chip select for the touch screen in the TFT_eSPI library.  Adafruit have a tutorial that supports you display so you should be able to adapt the examples.

I do not have a display with the STMPE610 controller so am unable to help.

closed 2018-07-25 by Bodmer

wtilton, 2018-07-26:
Ok good to know. I'll comment the TOUCH_CS out in the User_Setup and use Adafruit's library. Hope this helps someone either way. Thanks again.