Bodmer/TFT_eSPI #175: Anyone make ili9488 SPI with touch working?

Mike-the-one, 2018-07-26:
We have an 3.5inch LCD, ili9488 and xpt2046 in SPI mode, with the updated lib, LCD works, but cannot get touch to work.

Anyone tried that?  Does xpt2046 touch work in latest lib with ili9488 in spi mode?

Bodmer, 2018-07-26:
Please provide a link to the display you have so I can check compatibility.

Are you using ESP8266 or ESP32?

How have you connected it up?

jiangminghe, 2018-07-28:
Hi,Bodmer.
My name is Minghe from China.
I tried with your files for ILI9488.
The LCD displays works well, but touch doesn't work.
As you mentioned, i tried with Test_Touch_Controller.
And the output serial monitor displays in real time "X=0 Y=0 Z=4096"

Bodmer, 2018-07-28:
@Mike-the-one 

How have you connected the XPT2046 to the ESP32?

@jiangminghe Please raise a new issue to avoid solution clashes

Mike-the-one, 2018-07-28:
@Bodmer Jiangminghe is working with me

@jiangminghe Could you please paste the connection?

jiangminghe, 2018-07-28:
I connected as the followings

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    15  
#define TFT_DC    2  
#define TFT_RST   4
#define TOUCH_CS 21

Mike-the-one, 2018-07-28:
@Bodmer I think it is this:

```sh
#define TFT_MOSI  4
#define TFT_MISO 34
#define TFT_SCLK  2 
#define TFT_CS    5
#define TFT_DC   14
#define TFT_RST  12
#define TOUCH_CS 15
```

So like this:

```sh
Function          XPT2046    ESP32
MOSI               SDI (14)     4
MISO               SDO (12)   34
SCLK                SCL (16)    2
TOUCH_CS       CS (15)      15
```

Bodmer, 2018-07-28:
You have redefined pins 19 and 23 from the normal setup header files I use, so if you use your own pin allocations then this means the Touch SDI should connect to MOSI ESP32 pin 19, and SDO to MISO ESP32 pin 23:

Function          XPT2046    ESP32
MOSI              SDI (14)     19
MISO              SDO (12)     23
SCLK              SCL (16)     18
TOUCH_CS          CS  (15)     21

Bodmer, 2018-08-02:
Discussion moved to #177

closed 2018-08-02 by Bodmer