Bodmer/TFT_eSPI #176: ILI9341: Allowing a different read frequency for SPI may allow overclocking without (many?) artifacts

andrebraga, 2018-07-27:
Hi,

I realised that the issue with display garbling on graphicstest have a lot to do (if not entirely) with the read frequency set on SPI transactions. In other words it's hellishly more difficult for the ILI9341 controller to be overclocked on reads than on writes. I modified `spi_begin()` to take a default parameter (`spi_begin(boolean read = false)`) and inside of it I just hardcoded the SPI read freq:

```  if (locked) {locked = false; SPI.beginTransaction(SPISettings(read ? 20000000 : SPI_FREQUENCY, MSBFIRST, TFT_SPI_MODE));}```

(20MHz was the "sweet spot"; 27MHz led to color garbling but every pixel was in the right position.)

On every `read` function I simply replaced the existing ``spi_begin()` call to `spi_begin(true)`.

This allowed the `SOFTWARE SCROLL` tests of MCUFRIEND_kbv (which I just copied to another Arduino sketch that calls your lib) to work just right.

However, I'm wondering if instead of hacking the lib this way, would you be interested in allowing setting different SPI read and write freqs as constants on the header file? Or maybe just change them as a fraction of the max freq when they're higher than 27MHz? Or, conversely, making read and write freqs the same as long as they're below 20MHz, but lock it to 20MHz if write freq is above it?

Bodmer, 2018-07-28:
As you found, typically the read speed is slower than the write speed.  On the data sheet for the ILI9241 the maximum recommended write SPI frequency is 10MHz and 6.667MHx for a read, so a read is about 2/3rds of the speed.

On my display I start to get read errors at 40MHz and a few write errors at 80MHz.

I will update the library with an option to set the write speed at the next release.

Bodmer, 2018-07-29:
I have added an option for a different TFT read SPI frequency. Thican be specified in your setup file:

#define SPI_READ_FREQUENCY  20000000 // Optional reduced SPI frequency for reading TFT

closed 2018-07-29 by Bodmer

andrebraga, 2018-07-29:
Your approach is cleaner as despite allowing default values C++ doesn't have named parameters, but that led to some code duplication. Would you like a PR that refactors the code to the same functions except for the frequency setup? Since they're inlined the net gain is in maintainability.

andrebraga, 2018-07-30:
And oh, you missed the read freq on read settings struct and method.

(Is it really necessary to restore the write frequency after the read transaction guards? It's being reset on the transaction; maybe it's just better to set it there unconditionally if transactions are unavailable or disabled?)