Bodmer/TFT_eSPI #177: ESP32 with SPI ILI9488 Touch LCD

jiangminghe, 2018-07-28:
@Bodmer Thanks for your reply.
I connected ILI9488 with ESP32 as your user_setup.h

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    15 
#define TFT_DC    2  
#define TFT_RST   4
#define TOUCH_CS 21

Bodmer, 2018-07-28:
Have you connected up the touch controller SPI pins as well as the TFT?

Bodmer, 2018-07-28:
Are you sure the display as an XPT2046 controller fitted?

jiangminghe, 2018-07-29:
@Bodmer Thanks for your reply.
I will check again and come back soon.
Spend a good weekend.

jiangminghe, 2018-07-29:
@Bodmer  I have found out the main reason.
The main reason was to use the touch pad SPI as well as LCD in the same time.
Our pin-map was correct, also LCD and touch pad was working well individually.
But when we use LCD and Touch pad together, only LCD works.
I think you can solve this problem easily.
Thank you!

Mike-the-one, 2018-07-29:
@jiangminghe So if do the following, only touch works, LCD doesn't work:

```sh
xpt2046(14, Din/SDI) and  LCD(SDO) connect to ESP32 (MISO), 
xpt2046(12, Dout/SDO) and LCD(SDI) connect to ESP32 (MOSI),
```

And the following make LCD works but not touch:

```sh
Xxpt2046(12, Dout/SDO) and LCD(SDI) connect to ESP(MISO)
xpt2046(14, Din/SDI) and  LCD(SDO) connect to ESP(MOSI)
```

Correct?

jiangminghe, 2018-07-29:
Yes, That's it.
But when we connect as the followings, only LCD works.

xpt2046(14, Din/SDI)  and  LCD(SDI) connect to ESP32 (MISO-19), 
xpt2046(12, Dout/SDO) and LCD(SD0) connect to ESP32 (MOSI-23)

pbecchi, 2018-07-30:
SDO of lcd and touch  should go together!

jiangminghe, 2018-07-30:
Thank you for your reply.
I thought as you did.
Can you tell me exactly in code line?

pbecchi, 2018-07-30:
I mean the chip select CS HW lines .......!
As far as code...have you tried to reduce SPI speed  with lower #define SPI_FREQUENCY?

Bodmer, 2018-07-30:
The only wiring that should work is:
xpt2046(14, Dout/SDO) and LCD(SDO) connect to ESP32 (MISO-19),
xpt2046(12, Din/SDI) and LCD(SDI) connect to ESP32 (MOSI-23)

So it is peculiar that any other combination works at all!

Let me experiment by connecting up an ILI9488 and XPT2046. I will report back in a few days.

Which sketch are you using to test the display and touch controller?

jiangminghe, 2018-07-31:
@Bodmer Thanks for your immediate reply.
I used Test_Touch_Controller sketch for the touch test.
I will wait your experiment report.
Best regards!

pbecchi, 2018-07-31:
Since Bodmer comment is rigth, output of slave is input of master.... But it is easy to mix up input with output lines!
The only way is to test LCD and touch separately and this way the line connected to Miso is the slave output SDO.

BTW Lcd should work with just SDi.....SDO should be unused...at least for most chips.
It may be interesting to try disconnecting LCD SDO.

jiangminghe, 2018-07-31:
@Bodmer Sorry there was some confusion for pin-map.
I uploaded again the pin-map that shows the working state.
[image]
Please check and give me any advice.

jiangminghe, 2018-07-31:
@Bodmer I can use both LCD and touch when i cut down the SDO line of LCD.
I think you can modify your code easily.
I will wait your reply.
Thank you.
Best regards!

pbecchi, 2018-07-31:
Make sure you don't have problems with touch calibration....you may have to delete existing calibration file to make touch work! 
Anyhow it is good that now both touch and LCD works together ( with or without LcdSDO connected?).

jiangminghe, 2018-07-31:
@pbecchi Thanks for your reply.
I mean when i connect the SDO(LCD) with ESP32, then only LCD works (touch doesn't work.)
But when i disconnect the SDO(LCD) with ESP32, then both LCD and touch works.
so i think you need some modification in the code.
Please let me know your opinion.
Best regards!

pbecchi, 2018-07-31:
My previous comments was on Touch Controller Demo...make sure you recalibrate touch!
Regarding SDO Tft line:
With other chip controller like Ili9486 the SDO Hw line is not connected, so we may have to do it also here!
@Mike-the-one consider a possible interface HW modification !

Mike-the-one, 2018-08-01:
@pbecchi we can just cut the SDO line for now.

@jiangminghe Thanks!

@Bodmer Yes, I can confirm that by disconnecting SDO to LCD, both touch and LCD work.

pbecchi, 2018-08-01:
@Bodmer ..since cutting LCD SDO solve the problem, I guess this is due to ILI9488 communication to the Mcu. There should be a way to avoid those messages back from ILI9488, may be with a specific init command!

Bodmer, 2018-08-02:
I have created a test setup and have found that the SDO line stick low and does not go tristate even is the TFT chip select is high. This means that the SDO line must not be connected up if the SPI bus is to be shared with other devices. Disconnecting the SDO line also means that the TFT read funcions will no longer operate correctly.

The examples work fine and no changes to the library code are necessary. I will add a note to the ILI9488 setup header file.

closed 2018-08-02 by Bodmer