Bodmer/TFT_eSPI #180: Arduino Library manager does not pick up version changes to library.properties on GitHub

wilhelmzeuschner, 2018-07-31:
I cant update the library via the integrated library manager because it does not pick up changes made to library.properties here on GitHub.
I tried:
- lowering the version (even further) manually in my local file (no effect)
- changing the `url=` locally to https://github.com/Bodmer/TFT_eSPI (no effect)

I installed the library by including the .zip archive which i downloaded from GitHub... This might be the cause of this problem. 😕 
Since this method does not cause issues with other libraries, I post this issue anyways.

per1234, 2018-07-31:
>I cant update the library via the integrated library manager because it does not pick up changes made to library.properties here on GitHub.

No, it's because this library was never added to the Library Manager index so no matter what tricks you try you won't be able to install it via Library Manager.

If Bodmer wants the library to be added to the Library Manager index, they only need to open an issue here:
https://github.com/arduino/Arduino/issues
with a link to this repository.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

wilhelmzeuschner, 2018-07-31:
@per1234 Thanks for the info.
Since Bodmer keeps updating the version, it would make sense for the library to be included in the list.
I'll leave open the issue, maybe Bodmer decides to do so or responds. 
I'm very grateful for the work anyways!

Bodmer, 2018-08-02:
A request for inclusion in the library manager has been made. It is now up to the Arduino team.

closed 2018-08-02 by Bodmer