Bodmer/TFT_eSPI #184: Question: Using this with Oddwires IoT ESP32 and their IOT-BUS QVGA 2.4" TFT Touch Display

IraSch, 2018-08-04:
Hi,

I'm trying to use this with OddWires's IoT ESP32 and their 2.4" (320x240) TFT Touch Screen. 

All I get is a blank screen. My guess is that I don't have the correct DC and CS pins set somewhere. 

If I try Adafruit's feather library and examples, I get a white flashing screen but nothing more (using CS=15 and DC=33). 

The OddWires write-up on their display says that any ILI9341 Arduino library that supports the ESP32 may be used, and specifically recommends Bodmer's great TFT_eSPI library. Do you happen to know the correct DC and CS pins for their setup? Thanks.

Bodmer, 2018-08-04:
The [schematic](http://www.oddwires.com/iot-bus-qvga-2-4-tft-touch-display/) shows the pins to use. Assuming this is the display you are using then the pins are:

DC = GPIO 27
TFT_CS = GPIO 5

Bodmer, 2018-08-04:
Touch controller chip select is GPIO 16 so this needs to be defined too to avoid bus contention.

Bodmer, 2018-08-05:
_Do I just edit the lines in User_Setup.h following

 // For ESP32 Dev board (only tested with ILI9341 display)_

**Yes**

_Since GPIO16 is used for touch controller CS, do I need to use

#define D0_USED_FOR_CS_
**No, that is only for ESP8266**

Yes, you need to set GPIO 33 high in your sketch to enable the back light. TFT_eSPI does not do that for you.

Hope that helps.

IraSch, 2018-08-05:
It does. The 320x240 example sketches are working. 

The touch screen testsketch is a little strange. 

I'm getting good y values in the range of approx 200-3800, which based on previous experience with Adafruit's TFT display seems good. When the screen isn't pressed, y is pretty stable at 4095 - again, expected. 

X values seem odd. I either get values in the 4000's or 1900-2000 range and x doesn't follow the finger press as does y. X just seems to pick a number in either of those ranges, and kind of stays there even after no longer pressing the screen. 

Z values are typically in the 4100's when there is no press, and sometimes in the 6000's when pressed. I never cared about Z values so I'm not concerned here. With the Adafruit library, there was a way to see if someone pressed the screen (and then read values). I believe I found a similar library call with your "touch" library. 

Do I have some setting wrong that's causing the strange results for the X axis, or am I seeing a hardware issue? 

Thanks for all of your help. I'm getting close here.

Bodmer, 2018-08-07:
The communication with the touch screen seems to be working but the readings are wrong.

The screen can be wired up in different ways so the raw values reported can be high (~3800) or low (~400) when the screen is not being touched.  In all cases the Z (pressure) reading should be low when the screen in not being touched.  Here are some example raw value readings:

x: 416 y: 3741 z: 23
x: 416 y: 3711 z: 21
x: 1937 y: 2480 z: 1470 <<< touched here
x: 1952 y: 2492 z: 1774

I have had another look at the schematic and it does not show pin 10 as being connected to 3.3V, of course it may actually be connected on the board you have even though the drawing is wrong.  Pin 9 and 10 of the chip should be connected to 3.3V, maybe you can examine the board and see if that is the case.

Other than that I am not sure what is happening, the high y reading of 4095 suggests the value is wrong.

x: 416 y: 3741 z: 23
x: 416 y: 3711 z: 21
x: 1937 y: 2480 z: 1470 <<< touched here
x: 1952 y: 2492 z: 1774

IraSch, 2018-08-08:
I found the problem with the touch screen, and it was hardware. The ribbon cable from the TFT screen to the PCB wasn’t fully locked in. It's a pretty wide connector, so apparently some of the middle conductor weren't making contact. Once I locked it in, all is well. 

The touch controller chip on this TFT display has an IRQ line.  Since I didn't see a #DEFINE pin designation for the IRQ line, I'm guessing that your library polls the chip instead of using the IRQ (which is fine). If that's true, I may be able to cut the line on the board and gain another GPIO. 

Thanks for all of your help. Feel free to close out this thread (unless I need to do so as the originator).

Bodmer, 2018-08-08:
Thanks for the update.

As you surmised the library does not use the IRQ line.

closed 2018-08-08 by Bodmer