Bodmer/TFT_eSPI #186: LOLIN D32 Pro - conflicting pin names

perlix, 2018-08-10:
The new LOLIN D32 PRO board can now be selected in the Arduino IDE (after adding https://dl.espressif.com/dl/package_esp32_index.json to the Additional Boards Manager URLs). However, in the board's **_pins_arduino.h_** file, Espressif chose to declare TFT pin variables (for the on board TFT connector) that conflict with pin names defined in the TFT_eSPI library: 

```
static const uint8_t TFT_CS = 14;  // TFT CS pin
static const uint8_t TFT_LED = 32; // TFT backlight control pin
static const uint8_t TFT_RST = 33; // TFT reset pin
static const uint8_t TFT_DC = 27;  // TFT DC pin
```
There may be more elegant solutions, but I had to comment out these lines in _pins_arduino.h_ in order to make my existing sketches compile and run again (HX8357D display over parallel interface, D32 PRO's card reader over SPI). Besides, I think there shouldn't be pin declarations in _pins_arduino_h_ files anyway (apart from LED_BUILTIN, perhaps).

Bodmer, 2018-08-11:
Thanks for reporting this, I agree. Have you also reported this to the Espressif team.

Bodmer, 2018-08-14:
Using the D32 rather than the D32 PRO variant board profile will probably fix this as the only difference I can see is the addition of the TFT pins in the PRO copy of the  pins_arduino.h

closed 2018-08-14 by Bodmer

tobozo, 2018-09-17:
[fixed](https://github.com/espressif/arduino-esp32/commit/18d260ec6e5617ee3ddbe810b47661b3e8c44516) in the ESP32 source tree