Bodmer/TFT_eSPI #189: max function apparently doesn't work on ESP32 and eSPI

jlgoasis, 2018-08-13:
When compile appear " max was not declared in this scope" error. Please can you check it?
Please find below simple example:

#include <SPI.h>
#include <TFT_eSPI.h> 

TFT_eSPI tft = TFT_eSPI();

void setup() {
  tft.init();
  tft.setRotation(0);
}

void loop() {
uint8_t value = max(0,  10);
}

pipi61, 2018-08-13:
max declared in ......\hardware\espressif\esp32\cores\esp32\Arduino.h
try insert:
#include <Arduino.h>

Bodmer, 2018-08-14:
See [this issue.](https://github.com/espressif/arduino-esp32/issues/1734)

Bodmer, 2018-08-14:
This is error not caused by the TFT_eSPI library. I see I added this to TFT_eSPI.h to circumvent the problem with min():

#ifndef min
  #define min(a,b) (((a) < (b)) ? (a) : (b))
#endif

So you could add this in your sketch:

#ifndef max
  #define max(a,b) (((a) > (b)) ? (a) : (b))
#endif

closed 2018-08-14 by Bodmer

jlgoasis, 2018-08-16:
Thank you Istvan

I'll try it.

All the best

Jacinto