Bodmer/TFT_eSPI #190: ILI9341 with XTP2066 touch screen problem

cybercatnet, 2018-08-17:
Hi, @Bodmer ! First, thank a lot for sharing this lib!
I'm using a TFT 2.8 SPI ILI9341 XTP2046 with NodeMCU and the problem is that if I get the touch to work, it only does it with a couple of touchs and it dont work anymore. I'm doing print of get_Touch()  and always marks me '5' when this happens. 
I tried the Test_Touch_Controller and in that case it shows me the following results:
x:0    y:0    z:4095 (infinitely)

Check the wires, add external power supply but there were no changes.

This is the pin in the User_Setup.h

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D4  // Data Command control pin
#define TFT_RST  PIN_D0  // Reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1  // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

#define TOUCH_CS PIN_D2     // Chip select pin (T_CS) of touch screen

And this is the wiring that I use

[image]

Can you help me with this?

Bodmer, 2018-08-17:
I would remove the RTC clock connections and try again.

cybercatnet, 2018-08-17:
The same thing happens, I tried the Test_Touch_Calibrate again and it gives me the same data.
It's like constantly touch.. Could it have been damaged? Thank you!

Bodmer, 2018-08-18:
Can you provide a picture of tge back of the tft board. I think it will have a regulator fitted and this the VCC line should be connected to 5V. If you post a picture I will check it.

Bodmer, 2018-08-27:
Try the latest library  have made  a small change.

Run the Test_Touch_Controller again.

cybercatnet, 2018-08-28:
Thanks for answering. I try it and I tell you!

closed 2018-08-31 by Bodmer