Bodmer/TFT_eSPI #192: Wemos D1Mini -- moving to bare ESP-12[E/F] and ch340g

glown, 2018-08-22:
I apologize for asking, but I haven't found an appropriate forum for my question.

I typically use  TFT-eSPI on a Wemos D1-mini with several different displays ranging from a 128x128 to a 480x320 RPi display.  I really like the flexibility of the TFT-eSPI library and the Wemos hardware.

I'm trying for a more slim/sleek look so I want to use the bare esp12e/f chip along with a slim CH340G device I have.  I can do the basics, like load a simple program (GPIO15/GPIO0 pulldown and GPIO2 pullup) and run a program (GPIO15 pulldown and GPIO0/GPIO2 pullup).

But when I add other connections (SCK, SDA, A0, RESET, CS) things come to a screetching halt.  This should be possible, right?  Perhaps I'm missing something obvious? Does anyone have a project they can point at which used TFT-eSPI and a bare esp-12[e/f]?

Bodmer, 2018-08-22:
Which pins are you using for SCK, SDA, A0, RESET, CS.  How are you powering the setup?

glown, 2018-08-23:
Power is being supplied with a CH340g device (pictures attached).
[image]
These are the connections I use (wemos on left for reference)

```
CH340G         Wemos              ESP-12E                   128x128 Display

                                                                                 LED    ----| (wired to VCC)
    D5                     GPIO14                    SCK         |     
     D7                     GPIO13                    SDA         |
                      D3                     GPIO3                      A0           |
                      RST                   RST                           RESET     |
                      D8                    GPIO15                    CS            |
GND              GND                 GND                        GND        |
VCC                3v3                  VCC                         VCC     ---|
TX                                            RX
RX                                            TX

```

````

Bodmer, 2018-08-23:
I doubt the CH340G can supply the current needed for the ESP8266 and the display, plus you will need a decoupling capacitor on the supply as the ESP8266, maybe 100uF

glown, 2018-08-23:
I hadn't thought of that.  Made the assumption that since the I could drive the display from the wemos the ch340g could do it too.  I'll use a different power supply and look at the decoupling capacitor.  Thank you!