Bodmer/TFT_eSPI #194: Please add support for the ST7789 IPS 240x240 Display

wilhelmzeuschner, 2018-08-24:
Currently the library does not support displays with the ST7789 Driver chip that have a resolution of 240x240 pixels.

When uploading the rainbow example sketch, only a portion of the screen is used, even if 
```
#define TFT_WIDTH  240
#define TFT_HEIGHT 240
```
are set.

This is the display: https://www.aliexpress.com/item/1-3-inch-IPS-HD-TFT-ST7789-Drive-IC-240-240-SPI-Communication-3-3V-Voltage/32880444276.html

Thanks in advance!

Bodmer, 2018-08-26:
Support for the ST7789 with 240x240 display was added recently, but some screens have offsets that need to be accomodated. You should find that there is a rotation value that looks OK.

Post an image of your screen in rotation 0 so I can see if an offset is the problem.

There are no examples specific to  240x240 display so you may get a gabled screen display if you use the 320x240 examples.

wilhelmzeuschner, 2018-08-28:
Sorry for my late answer, particularly since it was me who opened this issue in the first place.

It is indeed possible to use this screen with the library as-is with no issues.
The examples just won't work as explained by you.
Here's one image showing the screen working: [image]

closed 2018-08-28 by wilhelmzeuschner