Bodmer/TFT_eSPI #195: LOLIN D32-PRO /w 128x128 ST7735S 1.4'' v1.0.0

tobozo, 2018-08-25:
Hello and thanks for this awesome library!

Since I've been struggling to find the right settings I guess it's worth sharing my setup and pin values:

- Lolin D32 Pro
- TFT 1.4 V1.0.0 (wemos.cc)

The D32-Pro has a [built-in socket](https://www.cnx-software.com/2018/06/15/wemos-lolin-d32-pro-esp32-tft-display-sd-card/) to match a [specific model of TFT](https://wiki.wemos.cc/products:d1_mini_shields:tft_1.4_shield), would that justify the creation of an additional file in the  user setups folder? (I can make a PR if necessary)

<div align="center" width="100%">
[image]

[image]
</div>

I haven't found a cleaner way to override default user setup in a persistent fashion (not deleted when I update TFT_eSPI) than defining these before including the library:
    
    #define ST7735_DRIVER
    #define TFT_WIDTH  128
    #define TFT_HEIGHT 128
    #define ST7735_GREENTAB2 // seems to be the more stable
    #define TFT_MISO 19
    #define TFT_MOSI 23
    #define TFT_SCLK 18
    #define TFT_CS    14
    #define TFT_DC    27
    #define TFT_RST   33
    #define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
    #define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
    #define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
    #define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
    #define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
    #define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
    #define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
    #define SMOOTH_FONT
    #define SPI_FREQUENCY  40000000
    // #define SPI_READ_FREQUENCY  20000000 // Optional reduced SPI frequency for reading TFT
    #define SPI_TOUCH_FREQUENCY  2500000
    #define USER_SETUP_LOADED // this prevents overloading from User_Setup

Bodmer, 2018-08-26:
Thanks for posting your settings. I have found the ST7735 will not work repiably at 40MHz but is OK at 27MHz.
Yes, you can post a setup file either as an attchement or a pull request.

tobozo, 2018-08-26:
here you go

[LolinD32Pro_ST7735S_128x128.h.txt](https://github.com/Bodmer/TFT_eSPI/files/2321728/LolinD32Pro_ST7735S_128x128.h.txt)

Bodmer, 2018-08-31:
Thanks I will add at the next update.

closed 2018-08-31 by Bodmer