Bodmer/TFT_eSPI #196: Getting random false values for x, y, and z on touch screen

IraSch, 2018-08-25:
Thank you for all of your help and this library. It is fantastic. 

I have an issue using the library with an OddWires ESP32 IoT board and their 2.4" TFT Touch Screen. The TFT continues to work perfectly, but the Touch returns false values. When nothing is touching the screen, Z values in the area of 2,000 or so are reported. When touching, they shoot up into the 3,000-3,500 range. x and y values don't particularly correspond to where the screen is touched (Raw, I understand). For example  X may be reported as 2xx and then 5xx for the same place. In some cases, the values tend to rise as you move your finger across the screen only to drop back and rise again. 

The TFT board is based on ILI9341 with a XPT2046 touch controller. I've also tried 2 TFT display boards and 2 ESP32 boards (to try to rule out a hardware issue in case the first TFT board itself was bad). 

My guess is that I have something set up incorrectly, but I believe my pin definitions are correct. I say this because I tried another touch controller library (FoxExe/STM32_XPT2046) before I found out that you already had "touch" supported directly by the TFT_eSPI library (should have figured that out on my own, but I didn't). That other library, when used by itself and no use of the TFT screen, reported x, y and z values correctly. But I couldn't run that library for touch and your library for TFT display - would get the same bogus touch responses. 

I've attached my user1 setup and sample touch screen sketch. Any help would be appreciated. 

Thank you!

[TouchFiles.zip](https://github.com/Bodmer/TFT_eSPI/files/2320781/TouchFiles.zip)

Bodmer, 2018-08-26:
Ryou setup looks OK, you have defined TFT_RST twice which will give a compile error but that is not the problem with the touch readings.

Run the Test_Touch_Controller sketch and post the results for a screen that in not touched and results for where the screen is touched in the middle.

IraSch, 2018-08-26:
Thanks for looking at this. Attached are two files. The first is running Test_Touch_Controller as requested. The first dozen or so lines are with nothing touching the screen. After that is touching the screen dead center. You'll notice that y readings while touching in the center vary from <10 to 9xx - this is something I see when running my finger across the screen - the numbers ramp up (still with random results), then drop back and ramp up again. I also see negative numbers from time to time. 

The second file is Test_Touch_Controller while running running my finger from bottom left to top right, with the screen in landscape orientation.

The third file is Read_User_Setup in case that gives any more info. 

NOTE: I've tried this with 2 TFT boards (one brand new out of the envelope that I bought thinking the first board was bad). Same results. I do not believe this is a problem with the TFT board, especially since the FoxEXE/STM32_XPT2046 library works just fine - but it doesn't talk to the TFT display at all - just the touch controller. Could it be some problem related to SPI interface (purely a guess)? 

As a possible added "clue", the TFT board also has an SD card slot (empty, not used), but according to the schematic, that is on SPI2 pins. 

Thanks Again, Ira

[TouchTests.zip](https://github.com/Bodmer/TFT_eSPI/files/2321636/TouchTests.zip)

Bodmer, 2018-08-26:
Can you run the Test_Touch_Controller again with the touch SPI frequency set lower at 1 MHz and post the output. e.g. use this in the setup file:

#define SPI_TOUCH_FREQUENCY  1000000

IraSch, 2018-08-26:
Looks much better. Attached is the file. The first section is no touch, followed by pressing in the center, and then moving from bottom left to top right. 

Unless I'm missing something, this seems pretty good now.

Any idea why this would be needed - or is this just what the correct setting for this TFT Touch board needs to be?

Thanks.

[Test_Touch_Controller_1MHz.zip](https://github.com/Bodmer/TFT_eSPI/files/2321933/Test_Touch_Controller_1MHz.zip)

Bodmer, 2018-08-26:
I suspect this is the screen settle time for the voltage, the signal needs to settle in 3 clock cycles and your screen has a longer time constant. I will think about how this can be accommodated in the library.

IraSch, 2018-08-27:
I replaced Touch.cpp. It still requires setting SPI touch frequency to 1MHz - the default of 2.5MHz doesn't work (and I'm not sure if it was intended to). 

Things are basically working now. I'm seeing slightly noisy results on the x axis, but not on y for some reason. Attached is a log generated by the touch test code while trying to hold a stylus still for a couple of seconds. You'll see a few erratic x points. This shows up much more clearly on a "paint" sketch (an example from Adafruit). Moving from left to right (landscape, axises flipped) shows a few points above and below the line. Moving up and down is clean (no spurious points to the left or right). 

I can certainly work with this - either by ignoring it since a 10% or so error in position isn't going to bother what I'm doing, or by averaging a few readings on a screen press and throwing out the extremes. 

At this point I'm happy, and can move forward. A BIG THANK YOU! for taking the time to address this. If you have an idea of what might be happening want to try something, I will certainly continue testing whatever you send. 

Once again, thank you!

[NoiseOnX.zip](https://github.com/Bodmer/TFT_eSPI/files/2325702/NoiseOnX.zip)

closed 2018-08-27 by Bodmer

IraSch, 2018-08-28:
It definitely does. Thank you for all of your help.