Bodmer/TFT_eSPI #197: TFT_eSPI class has no member "GetTouch Raw"

Uwe-HH, 2018-08-29:
Hello Bodmer,

I am a new member from Germany. I think I have a easy question -for you- but I could not find a solution.
Sorry that I disturb you today.

Since some weeks I used the ESP32 with the Arduino and I was very happy when I found your wonderful
TFT_eSPI Library. A great job, thank you so much !! Everything ist working good and I can use all your function to display text etc. on the ILI9341.

Now I changed to a TFT with touch function and I need your touch header and the souce file -touch.cpp. I found both in the Extension folder. But my sketch has no access to the information in these files.

I get the Error "class TFT_eSPI has no member named getTouchRaw.

I used only your example and there is the TFT_eSPI header included but not the touch.h  and also not the ...cpp.

What can I do, that the compiler knows that there is a touch.h and touch.cpp.

In  the last years I used only Microchip Mikrocontrollers and the Development System from Microchip, I have only less informations with the Arduino-System.

Thank you !!

Regards from Hamburg

Uwe-HH, 2018-08-29:
Thank you for the quick answer. 

The Error-Message is gone !!! I see I have to learn to read all your instructions before I ask for help!

closed 2018-08-31 by Bodmer

nguyencan99, 2020-10-30:
> You have to define the pin used for the [touch chip select](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L97) in the setup file (remove //) then the touch functions become available.

Thank you so much for this useful instruction!

tjross0308, 2023-10-09:
I am very sorry to do this. I am one of those "new ish" folks to all of this and I am having the same difficulty. I have done what I think is the correct process and commented in (removed the double //) so it looks like this :
 #define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

I did this under the section:
// For ESP32 Dev board (only tested with GC9A01 display)

I am still getting the error code listed above. 

I'm very sure that it is going to by very limited knowledge getting in the way, but if anyone has any suggestions I would be forever grateful.

Bodmer, 2023-10-09:
@tjross0308 
Which processor are you using?
Which interface to display, SPI or parallel?
Run the Read_User_Setup sketch and check the serial monitor output lists the touch pi n.