Bodmer/TFT_eSPI #198: Issue with Sprite in PSRam

wawibu, 2018-08-30:
Hi Bodmer,
I truly like your library and as I have recently switched to the ESP32 Wrover and with ILI9481 in parallel mode.
The Wrover offers a hugh amount of PSRam which I thought could be used to create a full screen sprite (480x320x16) in the memory. Writing it and push it afterwards to the TFT. To get the sprite created in PSRam I changed the calloc to ps_calloc (row 70, 83 and 106 within sprite.cpp). The sprite is now created successfull within the ini of the TFT and I have still 4,070,672 left as free heap.
I'm able to draw to the sprite within the memory and push it to the TFT. But sometimes the colors are shown incorrect. With the next refresh everything could be fine or other areas are shown in wrong colors.

Here the sun is only half in yellow:
[image]

On the next refresh the sun is full in yellow but now the moon at the bottom has the wrong color
[image]

Did you have an idea what goes wrong?
Shouldn't it be equal if the sprite is in ram or psram? 

Have also tried to create the sprite in 480x320x8 - but that has the same behavior. I'm wondered why it works only partly.

Any hint is appreciated.
Thanks,
Torsten

Bodmer, 2018-08-30:
Interesting, I did not know that the ESP32 Arduino environment supported PSRAM, so that is good news to me.

Some research on the web indicates that GPIO 16 and 17 are used for the PSRAM interface, are you using those signals for the interface to the TFT?  If so that may cause a problem.

I see there is a hardware bug in the ESP32 silicon and that there is a workaround for that in software that is built into the IDF. I am wondering if there is a problem still or if that bug fix has been migrated into the Arduino IDE.  It would be interesting to do a test to see if the problem is in the read process from PSRAM or the write process, or in the TFT write process.  If the problem is in the PSRAM write then the read would consistently show the error, so perhaps it is in the read? A checksum on the PSRAM read data would indicate if the problem is in the PSRAM interface.

I have an M5 Stack so I will try some tests, I suspect I will need to update the ESP32 environment if the PSRAM addition is recent.

It would be interesting to test the screen with colour bars rather than mostly black and white. This woul

wawibu, 2018-08-30:
I'm using PlatformIO with the following config:
`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_build.partitions = my_partition.csv
upload_port = 10.1.2.30

build_flags =
  -DBOARD_HAS_PSRAM
  -w
  -Os
  -DUSER_SETUP_LOADED=1
  -DILI9481_DRIVER=1
  -DLOAD_GFXFF=1
  -DESP32_PARALLEL=1
  -DTFT_CS=33
  -DTFT_DC=19
  -DTFT_RST=32
  -DTFT_WR=4
  -DTFT_RD=21
  -DTFT_D0=13
  -DTFT_D1=12
  -DTFT_D2=14
  -DTFT_D3=27
  -DTFT_D4=26
  -DTFT_D5=25
  -DTFT_D6=2
  -DTFT_D7=15
`
GPIO 16 / 17 are not available on the Wrover due to the fact that this is used for the PSRAM

I will do further full color test. 

Just updated the ESP32 platform to 1.3.0 and will test further

Bodmer, 2018-08-30:
It looks like occasionally the bytes in the weather (cloud/Sun) and Moon images are getting corrupted when they are read or copies to the Sprite. How are these images stored? Are you using pushImage() to copy these to the Sprite or to copy them direct to the TFT?

HumamHelfawi, 2019-03-22:
For people coming from google:
When you use Wrover with PSRAM, make sure to not use those pins:
6-11, 16, 17, 9, 3

This combination works for me:
#define ESP32_PARALLEL
#define TFT_CS   15  // Chip select control pin (library pulls permanently low
#define TFT_DC   2  // Data Command control pin - use a pin in the range 0-31
#define TFT_RST  -1  // Reset pin, toggles on startup

#define TFT_WR   4  // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD    -1  // Read strobe control pin  - use a pin in the range 0-31

#define TFT_D0   22  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2    26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   0  // TFT screen update performance.
#define TFT_D4   12
#define TFT_D5   21
#define TFT_D6   27
#define TFT_D7   14

chedersmith, 2023-05-29:
I liked the weather station could you pass me the code for my personal use