Bodmer/TFT_eSPI #199: Horizontal scrolling

abdollahpour, 2018-08-31:
Is it possible to have the horizontal scrolling? I did not find anything about it in the library

Bodmer, 2018-08-31:
I assume you know that screens natively support vertical scrolling in portrait orientation. This is used in the [example here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/320%20x%20240/TFT_Terminal).

Scrolling in Sprites, horizontally, vertically and diagonally is supported as in the appropriately names [examples here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Sprite).

closed 2018-08-31 by Bodmer

abdollahpour, 2018-09-01:
Thanks for the answer. I assumed hardware scrolling using without spirit would be faster and does not need extra memory. My device uses the LCD in portrait mode.