Bodmer/TFT_eSPI #202: Using GFXFF Fonts !

LongmontRobotics, 2018-09-04:
I have combined two Arduino projects to create a Running Timer on a TFT Display.
I have connected an Sparkfun RV1805C3 board to an ESP32 prototype board.
I have also connected a RasPi display and #define RPI_ILI9486_DRIVER.
The demo programs for this clock chip and display works well.

The RV1805C3 is programmed to output a 1 second pulse train and triggers an interrupt.
At each interrupt I increment a second and minute counter and display "MM:SS" on the Raspi display.

When I use an included Font(8) everything looks like it works.
When I try to use an GFXFF font, the display shows strange results.

I have reduced the program to a minimum set of functions, when a simple change to used the:
tft.setFreeFont(FSSBO24);

In the setup() function there are commented out lines for the FSSBO24 font.
Compiling those lines and adding GFXFF to the tft.drawchar and tft.drawnumber line in the main loop upsets the display.

Am I using the extra Fonts wrong ?
Thank You for looking into this.

/*
  RV-1805-C3 Set Date Time Example

  Configures date and time to the RTC module via the serial console, and then
  prints the current date and time in ISO 8601 format every second.

  Copyright (c) 2018 Macro Yau

  https://github.com/MacroYau/RV-1805-C3-Arduino-Library
*/
#include "Free_Fonts.h" // Include the header file attached to this sketch
#include <Arduino.h>
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

#include <RV1805C3.h>

const unsigned int tickInterruptPin = 16; // Connected to pin 3 of RV-1805-C3
const unsigned int rpmInterruptPin  = 12; // 

RV1805C3 rtc;

bool tickInterruptTriggered = false;
static uint32_t rpmInterruptTriggered  = 0;

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

static unit8_t hh, mm, ss;
static uint8_t Yr, Mo, Dy;

static uint32_t lastMillis = 0;
static uint32_t num_mills;
static uint32_t total_mills = 0;

static uint32_t rpmTimer     = 0;
static uint32_t old_rpmTimer = 0;
static uint32_t rpmTime      = 0;

static uint16_t upCountSec;
static  int padding;
void setup() {
	
	tft.init();
	
	tft.setRotation(1);
	tft.fillScreen(TFT_BLACK);
	
	tft.setTextSize(1);
	tft.setTextColor(TFT_YELLOW, TFT_BLACK);
	
	Serial.begin(115200);
	Serial.println();  Serial.println();
	Serial.println("tft.init()");
	Serial.println();  Serial.println();

	
//  Serial.println("RV-1805-C3 Set Date Time Example");
//  Serial.println();
//  Serial.println(__DATE__" : "__TIME__);
//  Serial.println();

  pinMode(tickInterruptPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(tickInterruptPin), tickInt, FALLING);

  pinMode(rpmInterruptPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(rpmInterruptPin), rpmInt, FALLING);

  Wire.begin();

  while (rtc.begin() == false) {
    Serial.println("Failed to detect RV-1805-C3!");
    delay(5000);
  }

  Serial.print("Start: ");
  Serial.println(rtc.getCurrentDateTime());

// SQW = 1 PPS
  rtc.writeByteToRegister(REG_SQUARE_WAVE, 0x8f);

// CLK/INT = SQWE
  rtc.writeByteToRegister(REG_CONTROL2, 0x01);

  Serial.println();

/*
  tft.setFreeFont(FSSBO24);
  tft.setTextDatum(ML_DATUM); 
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
 int padding = tft.textWidth("99:99 ", GFXFF); // get the width of the text in pixels
 tft.setTextPadding(padding);
*/
}

uint8_t _readDateTime[DATETIME_COMPONENTS];

uint8_t regConvertToDecimal(uint8_t bcd) {
	return (bcd / 16 * 10) + (bcd % 16);
}

static uint8_t secCntr = 0;
static uint8_t oldSecCntr = 99;
static uint8_t minCntr = 0;
static uint8_t oldMinCntr = 99;

void loop() {

	if( rpmInterruptTriggered )
	{
		rpmTime = old_rpmTimer - rpmTimer;
		if( rpmInterruptTriggered > 1 )
			rpmTime /= rpmInterruptTriggered;

		rpmInterruptTriggered = 0;

		static char rpmString[12];
		sprintf(rpmString, "RPM = %04d", rpmTime); 
		
		Serial.println(rpmString );
		Serial.println("");

	}

	if (tickInterruptTriggered) {
		tickInterruptTriggered = false;
		tft.setTextColor(TFT_WHITE, TFT_BLACK);

	 // Serial.print("Seconds : ");
	  Serial.println(rtc.getCurrentDateTime());

    // Update running digital time
	secCntr++;		// Advance Seconds Counter
	if( secCntr == 60 )
	{
		secCntr = 0;
		minCntr++;
		if( minCntr > 59 )
		{
			minCntr = 0;
		}
	}

	 rtc.readBytesFromRegisters(REG_TIME_HUNDREDTHS, _readDateTime, DATETIME_COMPONENTS);
		Yr = regConvertToDecimal(_readDateTime[DATETIME_YEAR]);
		Mo = regConvertToDecimal(_readDateTime[DATETIME_MONTH]);
		Dy = regConvertToDecimal(_readDateTime[DATETIME_DAY_OF_MONTH]);
		hh = regConvertToDecimal(_readDateTime[DATETIME_HOUR]);
		mm = regConvertToDecimal(_readDateTime[DATETIME_MINUTE]);
		ss = regConvertToDecimal(_readDateTime[DATETIME_SECOND]);

    int xpos = 100;
    int ypos = 50;
    int ysecs = ypos;

    if (oldMinCntr!= minCntr) // Redraw minutes counter every minute
	{ 
      oldMinCntr= minCntr;
      // Draw minutes
      if (minCntr < 10) xpos += tft.drawChar('0', xpos, ypos,8); // Add hours leading zero for 24 hr clock
      xpos += tft.drawNumber(minCntr, xpos, ypos, 8);             // Draw hours
      xpos += tft.drawChar(':', xpos, ypos, 8);
	xsecs = xpos; // Save seconds 'x' position for later display updates
    }
	  if (oldSecCntr != secCntr)
	  { // Redraw time every second
		  oldSecCntr = secCntr;
		  xpos = xsecs;
      if (secCntr< 10) xpos += tft.drawChar('0', xpos, ypos, 8); // Add minutes leading zero
     xpos += tft.drawNumber(secCntr, xpos, ypos, 8);             // Draw minutes

	}
}
}

void tickInt()
{
	tickInterruptTriggered = true;
}

void rpmInt()
{
	rpmInterruptTriggered++;
	old_rpmTimer = rpmTimer;
	rpmTimer = millis();
}

Bodmer, 2018-09-04:
Can you post a picture of the good and altered display.  Enable each of the commented out lines in turn to see which is affecting the display. Note tgat ypu are moving the datum that willl change the position of the numbers and setting a padding width based on thr GXFF font but always use font 8 for the numbers.

LongmontRobotics, 2018-09-04:
[TFT_Display.pdf](https://github.com/Bodmer/TFT_eSPI/files/2350251/TFT_Display.pdf)

I included the interrupt as it seems like the interrupt is causing this problem.
Still checking.

LongmontRobotics, 2018-09-05:
[image]

In setup(), I un-commentated just the  tft.setFreeFont(FSSBO24);

In loop() the line that prints the colon is off center even tho the ypos is the same as the other numbers.
This error causes the next char to be printed on the wrong line as well.
Without printing the colon, the line that prints two digits works on the correct ypos.

Bodmer, 2018-09-05:
Thanks, the problem is that drawChar uses s default datum to be compatible with the Adafruit_GFX library. Use drawString() instead, that should work.

LongmontRobotics, 2018-09-05:
[image]

Thank you

closed 2018-09-05 by LongmontRobotics