Bodmer/TFT_eSPI #203: Suggestion: animated GIF support

MarcFinns, 2018-09-06:
Hello Bodmer, a number of weather services provide animated GIFs with moving weather.
Do you think this is beyond the limits of esp8266/esp32?  would it be an interesting option to support it?
thanks.

Bodmer, 2018-09-06:
Have you tried[ this library](https://github.com/prenticedavid/AnimatedGIFs_SD)?

MarcFinns, 2018-09-06:
Well, i searched and missed this one... thanks!
Let’s see if it works. Need first to solve the out of memory as these services sadly all use https which crashes my device for now.
Thanks again!

closed 2018-09-10 by Bodmer

nikthefix, 2023-12-17:
Also this one:

https://github.com/bitbank2/AnimatedGIF

He has a TFT_eSPI example which works really well and can play back from PROGMEM.