Bodmer/TFT_eSPI #204: Sonar Radar with ILI8496 480x320

helinick81, 2018-09-07:
Hi all,

First things first.. I dont have an issue more the fact i am a NOOB! so struggling bad and i need someone to rescue me PLEASE !

I Am building a sonar radar, i got it working outputting data to Processing on the PC...  so far so good.
thinking i was a PRO and it was easy I decided to up the stakes and try and get my 4 inch TFT using ILI9486 480x320 screen to display the radar scan instead of the PC... 
now here is when things started going south :-( 
I found an example online [http://richmondsystems.net/2017/08/27/arduino-radar-using-arduino-nano-and-tft-screen/](url)

Note: I have only managed to get the TFT_sSPI to work on my ILI8496 screen... no other library.

but the gentleman who did this example is using TFT.h and spi.h libraries not the TFT_esPI.

i am trying to figure all this out and i got as far as drawing the radar outline on screen. ( took all day.. its a miracle i got this far to be honest - see attached pic)
[image]

now i got to the part of his code that i need to draw the radar sweep line that follows the servo angle... my head is going to explode for the past 4h trying to understand how to convert this to use the tft_eSPI library instead

`
tft.line( 80, 128, 80 + r_beam * cos( ( 360 - i ) * 3.14 / 180 ), 128 + r_beam * sin( ( 360 - i ) * 3.14 / 180 ) );`

i keep getting this error : TFT_eSPI::drawLine(double, double, int)'

can someone help me please... i am about to give up on it.

below is my code - yes i know its tragic in regards to layout and content.. way to much copy and paste ... 

`
#include <ESP32Servo.h>
#include <NewPing.h>
#include <WiFi.h>
#include <TFT_eSPI.h>
TFT_eSPI tft;
int distance;
const char *ssid      = "Inav";
const char *password  = "PZBZYLOH";
// Defines Tirg and Echo pins of the Ultrasonic Sensor
const int trigPin = 25;
const int echoPin = 26;
// Variables for the duration and the distance
long duration;

Servo myServo; // Creates a servo object for controlling the servo motor

char rc_Printout[ 4 ];
int interval = 0;

const int centreX  = 240; // Location of the Radar
const int centreY  = 320;
const int diameter = 238;  // Size of Radar

#define BLACK       0x0000
#define RED         0xF800
#define GREEN       0x07E0
#define BLUE        0x001F
#define CYAN        0x07FF
#define MAGENTA     0xF81F
#define YELLOW      0xFFE0
#define WHITE       0xFFFF
#define MAX_DISTANCE 300

void setup()
{
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an Output
  pinMode(echoPin, INPUT); // Sets the echoPin as an Input
  Serial.begin(115200);
  myServo.attach(27); // Defines on which pin is the servo motor attached
  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(BLACK);
  WiFi.begin(ssid, password);
  delay(1000);
  tft.print("Connected to: " + WiFi.SSID()); // Report which SSID has been used
  delay(1000);
  tft.print("  IP address: ");
  tft.print(WiFi.localIP());               // And the IP address assigned  display_status();
}
void loop() {
  tft.drawCircle(centreX, centreY, diameter, GREEN); // Draw Radar Scope outline

  // rotates the servo motor from 0 to 180 degrees
  for (int i = 0; i <= 180; i++) {
    myServo.write(i);
    delay(30);
    distance = calculateDistance();// Calls a function for calculating the distance measured by the Ultrasonic sensor for each degree

    Serial.print(i); // Sends the current degree into the Serial Port

    Serial.print(","); // Sends addition character right next to the previous value needed later in the Processing IDE for indexing

    Serial.print(distance); // Sends the distance value into the Serial Port

    Serial.print("."); // Sends addition character right next to the previous value needed later in the Processing IDE for indexing

  }

  // Repeats the previous lines from 180 to 0 degrees
  for (int i = 165; i > 15; i--) {
    myServo.write(i);
    delay(30);
    distance = calculateDistance();
    Serial.print(i);

    Serial.print(",");

    Serial.print(distance);

    Serial.print(".");

  }

}
// Function for calculating the distance measured by the Ultrasonic sensor
int calculateDistance() {

  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  // Sets the trigPin on HIGH state for 10 micro seconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  duration = pulseIn(echoPin, HIGH); // Reads the echoPin, returns the sound wave travel time in microseconds
  distance = duration * 0.034 / 2;
  return distance;

}
`

I have ZERO idea why it won't enter all the code in the current format on here.. it does partial but not all.. im in for a kicking... i can see it coming already. sorry guys.

Bodmer, 2018-09-08:
Hi, you have achieved a lot to get this far.  I am busy with work at the moment but towards the end of the week I will have some time free to help.  TFT_eSPI uses the same basic graphics calls as the Adafruit_GFX kibrary, so check out the Adafruit tutorial for that library.

Tge code will nit pasre as Github uses markup, so some character sequences change the formatting. It is better to put the code in a zip file and add that. However you link shows where the code has come from. The original sketch uses different graphics calls so they need to  echanged as you hsve surmised.

Bodmer, 2018-09-14:
I have adapted the code, there is still some work for you to do, see areas where I have put <<<< to mark test code changes.

[radar_v1.zip](https://github.com/Bodmer/TFT_eSPI/files/2384397/radar_v1.zip)

helinick81, 2018-09-14:
Huge thank you Bodmer 
ill start looking at it now 
keep you posted

helinick81, 2018-09-14:
[https://drive.google.com/open?id=1Or1TpChIJYtLqdOko06u9c2JFI9Hx3Et](url)

That worked Great Bodmer .. you are a GEMIOUS BRO!!! 

ill have a play with it and try polish it up now. 
attached a small video of it in action 
Cant thank you enough.

closed 2018-09-14 by helinick81