Bodmer/TFT_eSPI #205: Not working Display with ST7789V Driver

Tombo89, 2018-09-09:
Hello,
i have a few Tfts from Newheaven Displays that only support 0 or 16 Bit Parallel Interface. Sadly they do not support SPI. The have a included ST7789V Driver. I tried to get them working with an esp32 Devkit and your Library.

I adaptet the user_setup.h as follows

```
#define ST7789_DRIVER

#define ESP32_PARALLEL

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin  - use a pin in the range 0-31

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
```

But I cant get the Display to work. Am I missing somthing?

If you are interested here is a link to the Display.

[](http://www.newhavendisplay.com/nhd24240320cfcsxnf-p-9494.html)

Bodmer, 2018-09-09:
Use the 8 bit mode, I see from the datasheet that this mode actually uses DB8-15 of the display. Are those the pins you have tried?

Tombo89, 2018-09-09:
Hello, where you got this Information from? For me it says in the Datasheet for the Display pins DB0-7.
on Page 5 there is a wiering diagramm.

[](url)http://www.newhavendisplay.com/specs/NHD-2.4-240320CF-CSXN-F.pdf

Tombo89, 2018-09-09:
Hello,
i was a little bit wrong in my first post.
my pin definitions are as follow

```
#define TFT_CS   15  // Chip select control pin (library pulls permanently low
#define TFT_DC   5  // Data Command control pin - use a pin in the range 0-31
#define TFT_RST  18  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin  - use a pin in the range 0-31

#define TFT_D0   13  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   12  // so a single register write sets/clears all bits.
#define TFT_D2   14  // Pins can be randomly assigned, this does not affect
#define TFT_D3   27  // TFT screen update performance.
#define TFT_D4   26
#define TFT_D5   25
#define TFT_D6   23
#define TFT_D7   22
```

I hope they are ok. I tried it with DB0-7 and 8-15 on Display both with no result.

Bodmer, 2018-10-25:
Closing pending test results with new displays.

closed 2018-10-25 by Bodmer