Bodmer/TFT_eSPI #206: Wrong pins in setup ?

YordanYanakiev, 2018-09-12:
Hello.
I am trying to make this one
[image]
works with ESP32, but I've stuck on the setup described in here:

[https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup16_ILI9488_Parallel.h](url)

According to it the pins shall go in packs probably sequentially, but looking at the circuit it's not.
Please confirm that the pins numbers mentioned is the GPIO numbers mentioned in here :
[https://raw.githubusercontent.com/playelek/pinout-doit-32devkitv1/master/pinoutDOIT32devkitv1.png](url)

AKA : 

#define TFT_CS   33   - **means that GPIO33 ?**
#define TFT_DC   15  - **means that GPIO15 ?**
#define TFT_RST  32  - **means that GPIO32 ?**

#define TFT_WR    4  - **means that GPIO4 ?**
#define TFT_RD    2  - **means that GPIO2 ?**

#define TFT_D0   12 - **means that GPIO12 ?**
#define TFT_D1   13 - **means that GPIO13 ?**
#define TFT_D2   26 - **means that GPIO26 ?**
#define TFT_D3   25 - **means that GPIO25 ?**
#define TFT_D4   17 - **means that GPIO17 ?**
#define TFT_D5   16 - **means that GPIO16 ?**
#define TFT_D6   27 - **means that GPIO27 ?**
#define TFT_D7   14 - **means that GPIO14 ?**

Bodmer, 2018-09-12:
Yes pin number equals gpio number. Pins are selected for the UNO format ESP32 board. Keep wires very short if you are wiring it by hand.

closed 2018-09-12 by Bodmer

YordanYanakiev, 2018-09-12:
I am not sure what UNO format means. Please explain.

YordanYanakiev, 2018-09-12:
[image]
[image]

the other issue is this board, which turns out I should use - most of the pins which is required is occupied already ( for the SD card ) or missing at all - 16/17. Is there a way to use it with this board ?

(P.S. sorry for the booring and newbie questions ).

wawibu, 2018-09-13:
This is a ESP32 Wrover Board. Pin 16/17 therefore used for the PSRAM.
To use the Display in parallel mode you need to have at least to have 13 pins available on your board. 8 pins for D0 to D7 and DC, RST, WR, RD, CS. Important note is that these needs to be on GPIO 0 to 31

This is my config for ESP32 Wrover w/o SD Card and the ILI9481 TFT in parallel mode:
 
  -DUSER_SETUP_LOADED=1
  -DILI9481_DRIVER=1
  -DLOAD_GFXFF=1
  -DESP32_PARALLEL=1
  -DTFT_CS=21
  -DTFT_DC=25
  -DTFT_RST=18
  -DTFT_WR=19
  -DTFT_RD=26
  -DTFT_D0=27
  -DTFT_D1=5
  -DTFT_D2=14
  -DTFT_D3=4
  -DTFT_D4=12
  -DTFT_D5=2
  -DTFT_D6=13
  -DTFT_D7=15

As you see, you can select the pins however you like - only restriction is that they need to be in the rage of 0 to 31. My selection is based on my PCB Layout to prevent crossing wires. Moved assignments around during the routing to have all lines without any crossing.

YordanYanakiev, 2018-09-13:
AWESOME !
Thank You **Bodmer** and **wawibu** !
Going to try this and will report back !

Bodmer, 2018-09-14:
Your pin choice looks fine. RS and DC are the same control signal so you have made the right connection.
Run the Read_User_Setup sketch and check the serial output carefully to make sure the compiler is using the right settings.

As your board has PSRAM you may need to set this up in the IDE?

YordanYanakiev, 2018-09-14:
Seems like it was a bad idea to wire the power supply of this screen to the power from the ESP board.
It turns out that the ESP board is fried ( at least for sure the USB chipset ).

Bodmer, 2018-09-14:
It looks like you have a Lolin32 Pro board, this has built in current limiters so I am surprised it got fried as the 5V output comes from the USB port via the current limiter.  I see also that some users reported that the board overheats and that the [board has now been retired](https://wiki.wemos.cc/products:lolin32:lolin32_pro), but it is not clear if the board was retired for this reason. I use the cheap ESP32 boards off eBay which work well.

The simplest solution for those UNO format boards is to buy the UNO format ESP32 boards [like this one](https://www.ebay.co.uk/itm/ESP32-WiFi-Bluetooth-UNO-WeMos-D1-R32-4MB-Flash-CH340-Board-for-Arduino-UK/152975590787?hash=item239e0e4583:g:jF0AAOSwcxVayd9b).

YordanYanakiev, 2018-09-14:
Oh. I didn't know this. Thank You very much Bodmer. I am getting one of these then.