Bodmer/TFT_eSPI #207: GFX font question

rocketbob, 2018-09-13:
What's the procedure for generating a standard header file for a ttf font?

Bodmer, 2018-09-13:
The font format is one conceived by Adafruit, you have a few options:

1. If you use Linux or are able to setup Mingw under windows then you can use the [tool provided by Adafruit here](https://github.com/adafruit/Adafruit-GFX-Library/tree/master/fontconvert).

2. You can use an online converter [such as this one](http://oleddisplay.squix.ch/#/home).

3. Also see info I posted here.

closed 2018-09-13 by Bodmer

rocketbob, 2018-09-14:
Awesome, thank you!

marcschn, 2018-10-18:
Don't feel to open a new topic as it relates to @rocketbob question: what's the process for large fonts? both the adafruit tool and the online converter fail at large fonts (for example OpenSans >56pt).

Bodmer, 2018-10-18:
Adafruit chose to use a signed 8 bit value for the font size and coordinate parameters, this can overflow for large fonts and thus constrains the maximum font size.

marcschn, 2018-10-19:
Thanks for your quick reply. That's unfortunate. Do the same limitations apply to your vlw antialiased font implementation? I'd rather use some sort of font than have to generate single bitmaps for each char... :see_no_evil:

Bodmer, 2018-10-19:
Yes, there are limits also with smooth fonts but you should be able to use fonts up to ~200 pixels high. 8 bit values are used to save on memory space but making them 16 bit would only require minor changes to the library because the vlw font files themselves use 32 bit numbers!

The squix online converter for Adafruit fonts fails above a 100 point size, but works at 100.