Bodmer/TFT_eSPI #216: Smoothed Fonts

stufisher, 2018-10-09:
Hello, i'm making good use of this on an esp32 in conjunction with the Homie-esp8266 IoT framework. I've noticed that if you mix gfx and smoothed fonts you need to add:

`gfxFont = NULL;`
to line 101 of Smooth_font.cpp, inside the following function:
`void TFT_eSPI::loadFont(String fontName)`

otherwise spacing gets broken as it still thinks a gfx font is loaded.

Also, the reason i'm mixing gfx and smoothed fonts is that actually smoothed fonts take a long time to load and draw (~100ms, and worse if you need to say load 2-3 fonts), which is too slow when aiming for 20-30fps or so. Is there any chance we could have a font cache in RAM for devices such as the esp32 wrover that has a ridiculous amount of ram? I.e. you could load all fonts up front form spiffs, then read from ram on each draw.

Bodmer, 2018-10-10:
Hi, thanks. Can you provide a simple example sketch that shows the problem so I can investigate as I do not see why you fix apparently works.

Bodmer, 2018-10-10:
To answer you question about using PSRAM, yes using that for fonts and a screen buffer is something that I will be looking at. I already have full screen 320x240 Sprites working out of PSRAM and the font files could be loaded into PSRAM too, but it will need some code changes.

wawibu, 2018-10-10:
can confirm that using PSRAM for Sprites will work. Have a full screen sprite of 320x480 working in PSRAM without a problem so far.

stufisher, 2018-10-11:
If you read through draw glyph it applies a special padding for gfxfonts, it sets it to 101 or something. If you didnt set gfxFont to null and load a vlw then that spacing gets applied to the vlw font. 

I'll try and put together an example but it should be something along the lines of: (plot the text inverted on a white background so its easy to see the padding):

`sp->setFreeFont(&Lato_Regular7pt7b);
sp->loadFont("fonts/Lato-Regular14");
sp->setTextColor(TFT_BLACK, TFT_WHITE);
sp->drawString("test string", 0, 0);`

I am also already making use of full screen frame buffers in psram (multiple, allowing transitioning), really cool, thanks!

Bodmer, 2018-10-11:
Thanks, I did not understand what you meant by "spacing", I see now it is a "padding" issue.

I have uploaded a fix.

closed 2018-10-11 by Bodmer