Bodmer/TFT_eSPI #217: ESP latest git version and ili9341

MarcFinns, 2018-10-10:
I have a weird behaviour, with the display freezing after a few hours. I used to have that problem but I solved it by reducing the SPi frequency to 20mhz. It is now back after recompiling with the latest ESP git version. Can it be that it breaks something in the Spi clock management?

Bodmer, 2018-10-10:
By "freeze", do you mean the processor crashes or that the software is running normally but the display no longer updates?

MarcFinns, 2018-10-10:
The latter. All works but the display is frozen. The only way out is a power cycle. 
The display has reset connected to esp reset and CS always enabled (lack of pins)

Bodmer, 2018-10-10:
Possible causes:
1. Overheating display, check the ribbon end.
2. Power supply inadequate, tends to show if WiFi is in use
3. Long wires causing noise during SPI transfer
4. Someting on the SPI bus corrupting transfers
5. Sleep and wake cycles of the processor
Run an example like the graphics test and see if it freezes.

MarcFinns, 2018-10-10:
I agree on all these. However the project is complete and has been running for a year without hardware changes. 
My wild guess was that the spi speed settings in your config file could be bypassed. Can it be? Any way to check? Otherwise I will need to open it up and use the oscilloscope....

Bodmer, 2018-10-10:
A display freeze suggests it is getting duff commands. Duff data would show as a corrupted display image. It is difficult to advise without knowing more about the operating environment.  Does tft.begin() + display update in the loop() bring the display out of the freeze? Sometimes the display can lock up and then refuses a software reset and needs a hardware reset, this may happens when power glitches occur or corrupted commands.

If you enable transactions in the setup.h file then the SPI rate will be configured at the start of every function.

Bodmer, 2018-10-10:
P.S. Are and functions in you code using interrupts?

MarcFinns, 2018-10-10:
The display freezes but the image is not corrupted. It simply does not update anymore. Interestingly, the system is designed so that after a timeout the backlight is off and in that situation the display is not updated anymore. However, when I  operate the device again, the display is frozen. The begin() or init()functions dont reset it, a power cycle is needed. 
The core uses interrupts. I use the SoftwareSerial and a Geiger counter. 
Again, I am puzzled as I had solved the issue long ago. I might try to go back to ESP2.3.0 to test, as I believe that is the only thing I changed.
Thanks for your comments!

Bodmer, 2018-10-11:
Will close for now pending your investigation. I think the ILI9341 is quite susceptible to power glitches and spikes on control lines.

closed 2018-10-11 by Bodmer

MarcFinns, 2018-10-12:
Ok, i will investigate and report back if i find something interesting. Any reasons why you asked whether i use interrupts?

MarcFinns, 2018-10-18:
FYI - apparently the issue was caused by spiffs getting fragmented and slow. Not sure why. Cleaned the FS, used a different logic, no issues anymore with the display. Not sure why...

Bodmer, 2018-10-18:
Thanks for letting me know.  I have had problems with SPIFFS on occassion, it seems there are some bugs in the SPIFFS code that allow the FLASH to get corrupted. It is odd that the screen should freeze though.