Bodmer/TFT_eSPI #218: [Question] Pins to not connect ?

YordanYanakiev, 2018-10-10:
[image]
Hello.
It turns out my display is actually ILI9488
(using this setup #include <User_Setups/Setup16_ILI9488_Parallel.h> works like a charm).
Anyway I am trying to save pins for my other devices, and I am about to remove them permanently, so I wonder if the pins which I surrounded with red paint is actually needed to work the display or not ?
( I am not planing to use the SD card on the display module ).

Bodmer, 2018-10-11:
Yes, in addition you can disconnect LCD_RD and pull the LCD line high if you do not read from the LCD. You can also disconnect TFT_CS and pull that LCD line low.

closed 2018-10-11 by Bodmer

YordanYanakiev, 2018-10-11:
Thank You very much Bodmer!

hamzamahmood-lm, 2021-05-22:
Hello @YordanYanakiev , which controller were you using if you remember ? 🤞

Bodmer, 2021-05-22:
Setup 16 is for an ESP32.

YordanYanakiev, 2021-05-23:
> Hello @YordanYanakiev , which controller were you using if you remember ? 🤞

The microcontroller was ESP32