Bodmer/TFT_eSPI #221: ESP32 with ILI9341 (WROVER KIT) display only on RST

Aietes, 2018-10-17:
I'm trying to use this library with the official ESP WROVER KIT, it's an ESP32 with an ILI9341. I figured out the right PINs, I think. The example code for the display works in a way, that while the display stays dark, when I reset the board, I can see the drawn image on the screen for a second, until the restart happens. So I'm guessing the backlight or reset of the display driver is somehow wrong. Any experience or help on that?

Aietes, 2018-10-17:
I tested activating the backlight manually after tft.begin(), which works:
```
  pinMode(5, OUTPUT);
  digitalWrite(5, LOW);
```
It seems the library fails to activate the backlight in my constellation. The TFT_BL pin is set correctly but doesn't seem to get set to low.

Bodmer, 2018-10-17:
Hi, yes, at the moment the library only drives a backlight pin if the M5STACK option is chosen, that TFT requires the line high to switch on the backlight. In the WROOM kit the backlight control pin must be driven low to turn on the backlight LEDs.  If you do not sepcify a pin , then the backlight is biased ON.

I can add a setup file for the WROOM, which pins are used for the LCD? Is it an ILI9341 in the kit?

I will be adding PSRAM support for the anit-aliased (smooth) fonts to speed up rendering and for Sprites so full screen frame buffers will be supported for large screens.

Aietes, 2018-10-17:
Thank you for your response! The pins are below, for a WROVER KIT with ILI9341:

    -DILI9341_DRIVER=1
    -DTFT_WIDTH=240
    -DTFT_HEIGHT=320

    -DTFT_MISO=25
    -DTFT_MOSI=23
    -DTFT_SCLK=19
    -DTFT_CS=22
    -DTFT_DC=21
    -DTFT_RST=18
    -DTFT_BL=5

Les-A, 2018-11-03:
Can I also tag onto this issue for the Rover kit with ST7789 TFT - again TFT_BL pin is 5 and needs to be driven LOW after initialisation.  Other pin definitions, etc are:

[code]
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 7789
Display width  = 240 
Display height = 240 

MOSI    = D23 (GPIO 23)
MISO    = D25 (GPIO 25)
SCK     = D19 (GPIO 19)
TFT_CS   = D22 (GPIO 22)
TFT_DC   = D21 (GPIO 21)
TFT_RST  = D18 (GPIO 18)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
[/code]

closed 2019-01-06 by Bodmer