Bodmer/TFT_eSPI #222: After update to version 1.0.2 display stop works on the ESP32.

vitzaoral, 2018-10-18:
Hello.
I've updated the library from version _1.0.1_ to _1.0.2_ and the display has stopped working (it only shows a white screen). After downgrade back to _1.0.1_ everything works properly. I am using ESP32 WROOM Kit and ILI 9341 TFT LCD Touch display (SPI, 320x240) with wiring and settings described [here](https://github.com/vitzaoral/esp32_thermostat). Is there some bug or was changed something? Anyway, great library and work!

Bodmer, 2018-10-18:
The changes between 1.0.1 and 1.0.2 are minor and should not impact the display interface.  You do need to change the setup to suit your hardware when you upgrade.  There is a diagnostic example sketch "Read_User_Setup" that can be run, it prints the settings that the compiler picked up. If you run it after compiling with 1.0.1 and then again after compiling with 1.0.2 ther should be no difference in the output.

vitzaoral, 2018-10-18:
You are right, the update has overwritten my User_Setup.h file and I did not notice it. Now everything is OK after the update. I apologize and thank you for your help!

closed 2018-10-18 by vitzaoral