Bodmer/TFT_eSPI #223: Can I use anti-aliased fonts with the standard font functions?

Aietes, 2018-10-19:
No issue, just a question: Is it possible to use smooth fonts (aa) with the string functions, e.g. drawCentreString(...) or setTextDatum(...)?

And thank you for the great work, this library is great! 👍

Bodmer, 2018-10-19:
Yes, you can.  This is used to good effect in my new [weather library](https://github.com/Bodmer/DarkSkyWeather) example to align the text nicely.

Aietes, 2018-10-19:
I see, very cool, thank you!

Bodmer, 2018-10-19:
I see that the examples included do not adequately cover this aspect of using the fonts.

There are different methods that can be used to render the fonts to the screen. I will be creating some examples and will upload them this weekend.

I will leave this open until I have added new examples.

Bodmer, 2018-10-21:
OK, I have added four new examples showing methods of rendering smooth fonts to the screen. These are in "examples/Smooth Fonts".  These are written for a screen size of at least 320x240.

I will close this issue. Please open again if you find any problems with the examples.

closed 2018-10-21 by Bodmer