Bodmer/TFT_eSPI #224: Cannot compile without smooth font

MarkyAD, 2018-10-23:
drawString() needs an #ifdef SMOOTH_FONT. Undefining SMOOTH_FONT in user setup leads to symbol not declared error on "fontLoaded" because fontLoaded is in Extensions/Smooth_font.h and is only included if SMOOTH_FONT is defined.
```
TFT_eSPI/TFT_eSPI.cpp: In member function 'int16_t TFT_eSPI::drawString(const char*, int, int, int)':
TFT_eSPI/TFT_eSPI.cpp:4164:45: error: 'fontLoaded' was not declared in this scope
   bool freeFont = (font == 1 && gfxFont && !fontLoaded);

Bodmer, 2018-10-24:
Thanks for reporting this I will correct this soon.

Leave SMOOTH_FONT enabled as in default for the time being.

Bodmer, 2018-10-24:
Fixed, thanks for pointing out this bug.

wawibu, 2018-11-05:
just updated to 1.1.1 and the issue is still coming up. SMOOTH_FONT still needs to be defined as otherwise frontLoaded is not declared at 4163 in TFT_eSPI.cpp

Bodmer, 2018-11-05:
It looks like you do not have the version [with this commit](https://github.com/Bodmer/TFT_eSPI/commit/3d5ff73690edb48f29c1a46cc88bd4a41e0b4e33).  I think if you download the latest it should be fixed.  I may have forgotten to raise the issue number.

Bodmer, 2018-11-05:
Yep, forgot to raise the version number! Please try the latest.

closed 2018-11-05 by Bodmer

Bodmer, 2018-11-06:
I have raised the version again to 1.1.3 with support for the WROVER kit.

I have created a release at 1.1.3 and this should be available via the library manager within 24 hours.

wawibu, 2018-11-08:
would like to confirm that 1.1.3 is now available in PlatformIO and is fixed
Thanks again for your great support and this great library!