Bodmer/TFT_eSPI #225: Smooth Fonts exception on ESP32

ImpulseAdventure, 2018-10-24:
Hi Bodmer --

Thanks again for the fantastic library! All my testing so far with the library has been running very well, however I appear to be running into processor exceptions on the **Smooth_Font** demos. I'm wondering if you might have an idea what could be triggering this?

**Library**: TFT_eSPI from repo commit b93a40a (2018/10/23)
**Platform**: SparkFun ESP32 Thing
**Display**: Adafruit 2.8" 320x240 TFT Touch Shield for Arduino (manually wired)

**Outcome**:
- All examples in [Examples -> Smooth Fonts] cause an exception, apart from `alphaBlend_Test` (which doesn't load the Smooth Fonts).
- All other examples I have tried appear to be successful (incl. TFT_graphicstest_onelib, Free_Font_Demo, Sprite_scroll_8bit, etc.)

**Notes**
- I have uploaded the font files to SPIFFS using the **ESP32 filesystem uploader** (hash verified)
- Other config settings:
```
#define ILI9341_DRIVER
#define TFT_CS 12
#define TFT_DC 14
#define TFT_RST 27
#define TOUCH_CS 13
#define SPI_FREQUENCY 27000000
#define SPI_READ_FREQUENCY  20000000
```

**Output**:
From `Font_Demo_2`:
```
SPIFFS available!

Fonts found OK.
Guru Meditation Error: Core  1 panic'ed (StoreProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d3473  PS      : 0x00060930  A0      : 0x800d40f1  A1      : 0x3ffca670  
A2      : 0x3ffc224c  A3      : 0x0000ffff  A4      : 0x00000000  A5      : 0x001bfffc  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d3471  A9      : 0x3ffca650  
A10     : 0xffffffff  A11     : 0xffff0000  A12     : 0xffffffff  A13     : 0x3f408c9a  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x400d3473:0x3ffca670 0x400d40ee:0x3ffca690 0x400d0d17:0x3ffca6d0 0x400eeea1:0x3ffca700

Rebooting...
```
- According to the exception debugger:
```
Decoding 7 results
0x400d3473: TFT_eSPI::loadMetrics(unsigned short) at E:\...\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp line 1136
0x400d3473: TFT_eSPI::loadMetrics(unsigned short) at E:\...\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp line 1136
0x400d40ee: TFT_eSPI::loadFont(String) at E:\...\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp line 1136
0x400d0d17: loop() at E:\...\Arduino\libraries\TFT_eSPI-master\examples\Smooth Fonts\Font_Demo_2/Font_Demo_2.ino line 86
0x400eeea1: loopTask(void*) at E:\...\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 18 (discriminator 1)

```

Any pointers would be greatly appreciated.
thanks

Bodmer, 2018-10-24:
It appears that accessing SPIFFS whem loading the font is causing an issue. I have no explanation for this as the examples run OK on my setup.

Which version of the ESP32 board support package are you running?

Do the SPIFFS test examples run from the SPIFFS library?

Are you using the Arduino IDE, if so which version?

Bodmer, 2018-10-24:
Can you try uploading the font files to SPIFFS with a benign program like "blink" loaded and running, then upload the Font_Demo_2 demo sketch. The reason I ask is that today I had an ESP32 that was stuck in a "crashed" loop and in that state a SPIFFS upload appeared to create a bad SPIFFS image full of garbage, even though the files apparently existed.

ImpulseAdventure, 2018-10-24:
Thanks Bodmer for the followup.

With original installation:
- I originally had IDE version 1.8.5, and the ESP32 installed manually via the /hardware folder (prior to the board manager JSON availability).
- Test: SPIFFS/SPIFFS_Test: works successfully
- Test: upload Blink, run, SPIFFS Sketch Data Upload (from Font_Demo_2) - hash verified, Blink resumes (after Data Upload's automatic hard reset),manual reset for programming, upload Font_Demo_2. See exceptions.

I then updated the IDE to 1.8.7 and used the latest ESP32 board package (1.0.0) from https://dl.espressif.com/dl/package_esp32_index.json
- Test: TFT_eSPI: Font_Demo_2: Still encounter the same exception
```
Decoding 7 results
0x400d353f: TFT_eSPI::loadMetrics(unsigned short) at E:\...\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp line 1136
0x400d353f: TFT_eSPI::loadMetrics(unsigned short) at E:\...\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp line 1136
0x400d41ba: TFT_eSPI::loadFont(String) at E:\...\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp line 1136
0x400d0dd3: loop() at E:\...\Arduino\libraries\TFT_eSPI-master\examples\Smooth Fonts\Font_Demo_2/Font_Demo_2.ino line 86
0x400f0472: loopTask(void*) at C:\Users\...\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/main.cpp line 17 (discriminator 1)
```

ImpulseAdventure, 2018-10-25:
Thanks Bodmer for digging into this!

I just tested out the alternate version of `libspiffs.a` that you provided.
Good news, it worked! I am able to run the anti-aliased **Font_Demo** sketches without exceptions now.

Bodmer, 2018-10-25:
OK, that is good news.  It appears that an incompatibility crept into the ESP32 package and that the IDE upload does not create file images in the correct format. It is good to know what the root cause is and that the incompatibility in not within TFT_eSPI.

I have a recent installation of the Arduino IDE and ESP32 tools on a Win 10 laptop and that does not show the issue so **maybe** the problem in the ESP32 package introduced in Jan 2018 was fixed after circa May 2018? I will have to upgrade my main desktop PC to use the latest ESP32 tools and see what happens.

closed 2018-10-25 by Bodmer

ImpulseAdventure, 2018-10-25:
Hey @Bodmer -- unfortunately it looks like I am continuing to see the same exception with **Smooth Fonts** active. Not sure why I might have experienced a "good" run yesterday. I have repeated the testing (including the replacement `libspiffs.a`) and I am observing the same exception every time I use Smooth Fonts on the SparkFun ESP32 Thing.

Regarding your comment on the latest ESP32 package -- note that I have been using the latest (ie. > May 2018).

### Output from Font_Demo_2 ###
```
SPIFFS available!

Fonts found OK.
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3537  PS      : 0x00060930  A0      : 0x800d41b5  A1      : 0x3ffb1f10  
A2      : 0x3ffc241c  A3      : 0x0000ffff  A4      : 0x00000000  A5      : 0x001bfffc  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d3535  A9      : 0x3ffb1ef0  
A10     : 0xffffffff  A11     : 0xffff0000  A12     : 0xffffffff  A13     : 0x3f409d82  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d3537:0x3ffb1f10 0x400d41b2:0x3ffb1f30 0x400d0dcb:0x3ffb1f70 0x400ef902:0x3ffb1fa0

Rebooting...
```

Given that the `SPIFFS_Test` shows a pass, do you have any other ideas of how one might narrow this down? (perhaps some way to test your idea that the SPIFFS content was corrupt / returning a bad font size?)

When running `Print_Smooth_Font`, I see:
```
Initialisation done.

SPIFFS files found:
Listing directory: /
  /Final-Frontier-28.vlw             25287 bytes

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3217  PS      : 0x00060930  A0      : 0x800d3955  A1      : 0x3ffb1f00  
A2      : 0x3ffc241c  A3      : 0x0000ffff  A4      : 0x00000000  A5      : 0x001bfffc  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d3215  A9      : 0x3ffb1ee0  
A10     : 0xffffffff  A11     : 0xffff0000  A12     : 0xffffffff  A13     : 0x3f409c0e  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d3217:0x3ffb1f00 0x400d3952:0x3ffb1f20 0x400d0d01:0x3ffb1f60 0x400ef4ca:0x3ffb1fa0

Rebooting...
```

Thanks!

Bodmer, 2018-10-25:
Other reports about a corruped file in SPIFFS report that read/write from a sketch always work OK. This implies that the transfer of files from the IDE serial port to SFIFFS uses a different software routine. It is not clear to me whether the ESP32 running software that does this work is at fault or there is a problem at the IDE end where there is a protocol mismatch. So I think a sketch based test will always pass in your case. One way to test this is to upload a test file, say full or test from a document and then read it with a sketch and send the output to the serial port. Maybe corruption will be visible. I have other stuff to do so I will not be able to experiment and in any case my tests will always pass...

I had a brief look to see if there are other ways of tranferring files to SPIFFS, ie via Wifi OTA update bit I did not look for long and did not notice a solution.  Another option is to get the files via a http GET from a server and have the sketch copy the files to SPIFFS, I have fetched jpeg files from Google maps and the imgur server and placed them is SPIFFS, but that is not a convenient method for sketch examples.

I have a laptop installation the has esp32 core files datestamped July 2018 and uploads to SPIFFS work fine to my boards.

Bodmer, 2018-10-25:
I think you can reduce the FLASH speed in the IDE via m menu option, have you tried that?

cuonghv2010, 2019-02-12:
Hi Bodmer for great lib.

I got a problem with smooth font and SD card.

I save my smooth font in SD card and of  change the function loadfont too.

`  if (SD.exists(_gFontFilename) == false) {
    Serial.println("Font file " + fontName + " not found!");
    return;
  }

  fontFile = SD.open( _gFontFilename, "r");`

The problem is I need to load smooth font every time i update new value show on LCD.
I call this function todo that every 50 ms
`
void print_SLCD(String text, int x, int y, int align, uint16_t text_color, uint16_t text_background_color, String font) {
  tft.loadFont(font); // Load font
  if (tft.fontLoaded) {
    tft.setTextColor(text_color, text_background_color);
    tft.setCursor(x, y);
    tft.print(text);
    tft.unloadFont();
  }
}`

It work but some time it load font failse and reboot.

`Fonts found OK.
Guru Meditation Error: Core  1 panic'ed (StoreProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d3473  PS      : 0x00060930  A0      : 0x800d40f1  A1      : 0x3ffca670  
A2      : 0x3ffc224c  A3      : 0x0000ffff  A4      : 0x00000000  A5      : 0x001bfffc  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d3471  A9      : 0x3ffca650  
A10     : 0xffffffff  A11     : 0xffff0000  A12     : 0xffffffff  A13     : 0x3f408c9a  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x400d3473:0x3ffca670 0x400d40ee:0x3ffca690 0x400d0d17:0x3ffca6d0 0x400eeea1:0x3ffca700

Rebooting...`

Note:
Platform: iot express mkii ( ESP WROOM-32)
LCD: Adafruit 2.8" 320x240 TFT Touch Shield
IDE: Arduino 1.8.8

Setup:
#define ILI9341_DRIVER
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   25  // Chip select control pin
#define TFT_DC   17  // Data Command control pin
#define TFT_RST   26 // Reset pin (could connect to RST pin)
#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD 
#define LOAD_FONT8
#define SMOOTH_FONT

#define SPI_FREQUENCY 10000000
#define SPI_READ_FREQUENCY  10000000
#define SPI_TOUCH_FREQUENCY  2500000
#define SPI_SD_FREQUENCY  2000000

thanks

Bodmer, 2019-02-13:
When loadFont() is called it opens the file, this is not closed until unloadFont() is called. You should not need to keep opening the file. I suspect the processor crashes because a small amount of memlry is used each time you open thr files, so perhaps that is why the process or crashes?

I have never tried putting the fonts on SD CARD and eeping fonts on SD card is likely to make rendering slow so SPIFFS is the better options.

At some point I plan to allow an ESP32 to load fonts into PSRAM (if fitted) and that will make rendering much faster.