Bodmer/TFT_eSPI #226: M5Stack with ili9341 - How do you change brightness?

timtimsson, 2018-10-25:
How can i change brightness?
If i use the M5Stack Library i can do this..
Is it possible to integrate this?
[`M5.lcd.setBrightness(uint8_t brightness);`](https://github.com/m5stack/M5Stack/blob/master/src/M5Display.h)

Bodmer, 2018-10-25:
There is no function for this in the TFT_eSPI library. You could use a PWM function in your sketch to do this.

The M5Stack library has added a wrapper to the TFT_eSPI library at it's core and added a brightness control function. I will add this to my TODO list but it is not a priority at the moment!

closed 2019-01-06 by Bodmer

Sarah-C, 2021-09-09:
M5Display.cpp, line 30... this is the code that sets the brightness in the M5Stack library:
ledcWrite is a standard ESP32 function in the Arduino IDE.

```
#define BLK_PWM_CHANNEL 7 // LEDC_CHANNEL_7
.......
void M5Display::setBrightness(uint8_t brightness) {
  ledcWrite(BLK_PWM_CHANNEL, brightness);
}
```

Also ............. if using the M5Stack Core 2 (the touch-screen version with PSRAM) you can use this (because the M5Stack Core 2 library function doesn't currently work anyway!):
https://github.com/Sarah-C/M5Stack_Core2_ScreenBrightness

```

// lvl: 0:(backlight physically off), 20: full brightness (2.8 volts)

// Overdrive:
// This gives an extra 16 brightness levels ABOVE Core2's standard setting.
// lvl: 0:(backlight physically off), 36: full brightness (3.2 volts)

void core2Brightness(uint8_t lvl, bool overdrive = false) {
  // The backlight brightness is in steps of 25 in AXP192.cpp
  // calculation in SetDCVoltage: ( (voltage - 700) / 25 )
  // 2325 is the minimum "I can just about see a glow in a dark room" level of brightness.
  // 2800 is the value set by the AXP library as "standard" bright backlight.
  int v = lvl * 25 + 2300;

  // Clamp to safe values.
  if (v < 2300) v = 2300;
  if (overdrive) {
    if (v > 3200) v = 3200; // maximum of 3.2 volts, 3200 (uint8_t lvl  = 36) absolute max!
  } else {
    if (v > 2800) v = 2800; // maximum of 2.8 volts, 2800 (uint8_t lvl  = 20)
  }

  // Minimum brightness means turn off the LCD backlight.
  if (v == 2300) {
    // LED set to minimum brightness? Turn off.
    M5.Axp.SetDCDC3(false);
    return;
  } else {
    // Ensure backlight is on. (magic name = DCDC3)
    M5.Axp.SetDCDC3(true);
  }

  // Set the LCD backlight voltage. (magic number = 2)
  M5.Axp.SetDCVoltage(2, v);
}

```

Bodmer, 2021-09-14:
The solution is dependant on the module you have, for example:

M5Stack Basic Core module: has the backlight ON/OFF on GPIO 32 and can be PWM'ed in a user sketch.

M5Stack Core 2 module: has a power controller APZ192 chip that managed backlight brightness, this is on the ESP32 I2C bus and thus requires a specific library/driver to manage.

There are other M5Stack products (M5Stack Fire etc) but have not looked at the technical details on those so you will need to check.