Bodmer/TFT_eSPI #227: Streaks on display?

pryankster, 2018-10-26:
Display is this guy:  http://www.hiletgo.com/ProductDetail/2157911.html

(I think :-) It's an ST7735R (correct colors and no extra pixels around the edge when I use ST7735_GREENTAB2 in User_Setup.h.

However, the display flickers on update and is very "streak-y".  I have two of these displays, and they both have the same issue, so it could be a bad batch, or it could be a configuration error.

[image]

Any help or hints would be greatly appreciated :-)

-- pryankster

pryankster, 2018-10-26:
That made it worse.  Proportial fonts in the Graphics test are all screwed up. I don't see as much flickering, but the streaking is still there.

moononournation, 2018-10-26:
ESP8266 or ESP32?

Bodmer, 2018-10-26:
This looks like a power supply problem.
Have you connected the GND pin to the GND/0V pin on your ESP32 or ESP8266?  The picture suggests to me that it is not connected (GND is usually the pin next to the Vcc pin). When GND is not connected the display ends up stealing power from the digital pins and as these are toggling this makes the display flicker.

It looks like you have a line of garbage pixels on the left edge.

Bodmer, 2018-10-26:
OK, I would add an extra direct GND link near the others wires, the reason I suggest this is that I cannot see the wire link and some proto boards have a break halfway along the power rail... See the [red and blue line break here!](https://www.alibaba.com/product-detail/protoboard-830-Point-Solderless-Bread-Board_60423464507.html), this catches people out.

The display initialisation code varies depending on the display driver chip built into the display. This init code varies depending on the driver type.  The ST7735R is a very common chip but I have display board very similar to yours fitted with a different driver chip (ST7735S and S6D02A1).  The displays from China are cheap but sometimes the vendor quotes the wrong chip type and identical PCBs are fitted with different driver chip screens! So maybe experiment with other driver options.

I see on the vendors product page they offer demo software, it would be useful to get a copy to check if the initialisation code is different. This code is important to get right as it sets internal display voltages etc, so maybe that is the problem. I expect the displays are good, I have many different cheap displays and only ever had one duff one, and one cracked on arrival.

moononournation, 2018-10-26:
I have just tested everything fine:
[image]

moononournation, 2018-10-27:
Remember ST7735 cannot run normal at 40 MHz.
My connection details:
`ST7735 -> ESP32`
`GND    -> GND`
`Vcc    -> 3v3`
`NC`
`NC`
`NC`
`RESET  -> GPIO17`
`A0(D/C)-> GPIO16`
`SDA    -> GPIO23 (MOSI)`
`SCL    -> GPIO18 (CLK)`
`CS     -> GPIO5`
`SCK`
`MISO`
`MOSI`
`SD_CS`
`LED+   -> 3v3`
`LED-   -> GND`

Bodmer, 2018-10-27:
@moononournation thanks for your input. The timings reported by @pryankster indicate a similar clock rate to yours. I suspect you are not using the master branch of TFT_eSPI as your timings are different e.g filled RRects slower, so you results may not be comparable. The white plastic display bezel is different on @pryankster's display to your's and my displays of this exact PCB design and this is an indicator of a different TFT module supplier and potentially a different driver chip. If you try different (i.e. wrong)  drivers in the libray setup do you get the streaking?

moononournation, 2018-10-27:
@Bodmer yes, you are right. My branch is substantially tuning the ESP32 performance and at the same time fixed ST7735_GREENTAB missing code. So my code running at 27MHz can faster than master branch running at 40MHz.

Bodmer, 2018-10-27:
@moononournation  pull requests for bug fixes and enhancements are always welcome. Keep the changes as minimal as possible so they are clearly identifiable as I may need to refactor the code to avoid breaking compatibility with different display types (and to fit in with any unpublished updates and bug fixes). What effect does the wrong driver have on your display, any streaking effects?

moononournation, 2018-10-27:
Wrong driver only show wrong color, but running ST7735 at 40 MHz have similar streaking effects.

Bodmer, 2018-10-27:
Ok,  @pryankster  try a lower SPI clock rate. Run Read_User_Setup diagnostic example and look at serial output to check the compiler is picking up the settings you think it should. Post the output here if you are not sure it is right.

Bodmer, 2018-10-27:
@pryankster I thought I may be able to help by trying a very similar looking display that I have never used before. This proved interesting as the only driver that worked was the S6D02A1_DRIVER.  I have not used one of these displays since migrating projects to the ESP8266 and ESP32 so it was interesting to see how fast typical sketch screens get drawn on a smaller display. I also found this controller seems to run fine at 40MHz, but I have not expensively tested it:
[https://i.imgur.com/pfRRSRM.jpg](url)

@moononournation, the PDQ test does not reflect the typical performance seen in normal sketches. I don't think anyone would want to write every pixel on the screen with drawPixel, however it is easy to get a better than 10x performance improvement if you did by intelligent use of Sprites.   The replacement code for testPixels() is below if you wish to experiment. I expect this could be made into a 20x improvement with a simple raw line buffer but it simply is not worth it.

TFT_eSprite spr = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

uint32_t testPixels()
{
  int32_t w = tft.width();
  int32_t h = tft.height();

  uint32_t start = micros_start();

  spr.createSprite(tft.width(), 1);

  for (uint16_t y = 0; y < h; y++)
  {
    for (uint16_t x = 0; x < w; x++)
    {
      spr.drawPixel(x, 0, tft.color565(x<<3, y<<3, x*y));
    }
    spr.pushSprite(0, y);
  }
  
  return micros() - start;
}

moononournation, 2018-10-28:
@Bodmer yes I know ESP32 + PSRAM can draw much faster. My tuning on ESP32 is actually some sort of side effect of refactoring. When I try to add support of ILI9225 and 3 OLED in my hand, I found the code in ESP8266 and ESP32 is totally separated. I need to write same logic twice and ESP32 ironically run slower than ESP8266. So I refactored the code. After refactoring ESP32 enjoy the ESP8266 tuning and run much faster.

Bodmer, 2018-10-28:
@moononournation Ok, yes, the code separation can be irritating and hinder updates but it is deliberate to  isolate changes to the code and it is easy for me to copy and adapt existing known working code without breaking ESP8266 support. It also helps avoid a lot of #ifdef's in the early stages that make the code less readable. Back on topic, if you have ideas to help @pryankster get the display working then they will be appreciated.

 @pryankster I disconnected GND on my test display, it continued to work fine albeit with a very dim backlight, no streaking though. The streaking when the SPI rate is too high is quite different to yours. The vertical streaking is occuring where there are more bright pixels in a column, I think this indicates the internal bias voltages are not being programmed correctly because the driver selected is incompatible or not supported by the library.  I have a sketch that may be able to read the chip ID but it has not been tested on many dispkays, I will see if it works on mine, then post it.

The good news for you is I think the display is wired up correctly and not faulty so we just have to get the init code right. If the vendor supplies code to you as indicated on the product page I can copy that. Sadly some vendors in China usually do not provide sensible tech support or the correct init code as they have little or no technical knowledge of what they are selling, but the low price is attractive if some wasted time is tolerable. Some displays work reasonably well after reset without any init code as they have sensible register defaults, so I will add a "reset only" driver to the library, as that may help.

moononournation, 2018-10-28:
I think my LCD is the same batch as @pryankster . If it is not frequency problem, it should be wire connection problems.
@pryankster do you have further findings or testing result?

Bodmer, 2018-10-28:
I found another display that looks very similar, this one only works correctly with a ST7735_REDTAB option so I think it is a ST7735S. I suspect there are copied/clone chips that behave slightly differently on the market too which complicates matters as they do not conform to the expected data sheets. 
@moononournation  @pryankster's display has a subtly different white plastic bezzle design which in my experience often indicates a different display batch (but not always a different driver!)

Bodmer, 2018-10-28:
@moononournation  Zooming in to the pics at the end of the TFT screen shows a different circuit track arrangement on the "brown" ribbon coming form the COG, so it is definitely a different vendors screen fitted or a different manufacture batch/version. Of course it may still be the same driver.

Bodmer, 2018-10-28:
Just remembered [this dialogue](https://forum.arduino.cc/index.php?topic=406501.0) a few years ago, those libraries were for AVR processors but the discussion is relevant. Some displays are fitted with a IL9163 driver chip. TFT_eSPI has the option for that driver:

  #define ILI9163_DRIVER
and the Samsung chip:
  #define S6D02A1_DRIVER // Note: ILI9163 display works with this driver!

I will see if the chip ID reader sketch posted runs on an ESP32.

pryankster, 2018-11-02:
thanks for your help with this.  Sorry I've been incommunicado :-(  ...  I've been slogged down with the real world this week.    I was able to get the display playing nicely when I let the ESP32 drive the reset line instead of just tying reset to 3v3.