Bodmer/TFT_eSPI #228: Wemos D1 ESP32, TFT ILI9481

HI-POT, 2018-10-26:
Hello,
First of all, want to thank you for this great project which makes our lives easier :)
Nevertheless facing a strange issue you may be faced as well during tests. Have modified my board as explained. When all wired and uploaded whatever sketch LCD keeps running for just a couple of seconds and starts flickering and finishes with the white screen. Do you think that could be that modification´s causing this as original PCB paths from pins 35,34 and 36 are still connected to ESP32 as well as new pins 33,32,15. Once modification was removed behavior stayed the same. The board (have 3 of them) which was not touched by modification keeps running fine without any interruptions. Uploading same sketches to all of them with same parameters. Any hint will be helpful.

Below setup I used:
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = Parallel 
Display driver = 9481
Display width  = 320 
Display height = 480 

TFT_CS   = D33 (GPIO 33)
TFT_DC   = D15 (GPIO 15)
TFT_RST  = D32 (GPIO 32)

TFT_WR   = D4 (GPIO 4)
TFT_RD   = D2 (GPIO 2)

TFT_D0   = D12 (GPIO 12)
TFT_D1   = D13 (GPIO 13)
TFT_D2   = D26 (GPIO 26)
TFT_D3   = D25 (GPIO 25)
TFT_D4   = D17 (GPIO 17)
TFT_D5   = D16 (GPIO 16)
TFT_D6   = D27 (GPIO 27)
TFT_D7   = D14 (GPIO 14)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Bodmer, 2018-10-26:
Can you provide a link to the ESP32 board you are using, I searched for Wemos D1 ESP32 but that board is not an "UNO format" board that I expected.

Bodmer, 2018-10-26:
Ah, it is one of these (2nd picture down)?

Pins 35, 34 and 36 are all input and it is OK to connect digital out pins to them.

I had difficulty understanding exactly what works and does not work from your post. Is it the case you have 3 identical D1 R32 boards, all modified to link up the pins and only one works OK?

Which sketch are you using to test the boards?

Bodmer, 2018-10-26:
Can you also provide a link to the display vendors product page and confirm that you display (front and back) is identical in every detail to the one shown.

HI-POT, 2018-10-27:
Did more testing and seems out of 3 now 2 are running fine. Let them running over the night and the noise disappeared. 3rd one still white screen but seems suitable for other more simple projects. Looks like this issue is HW related as all components were bought on Aliexpress. Lesson learned... always buy more if it´s cheap :) 
1. Both running for almost 24hrs already, left side MCUFRIEND_kbv, sketch GLUE_Demo_480x320, 
right side TFT_eSPI, sketch TFT_Meters for resolution 480x320
https://drive.google.com/file/d/1PdqTzbGCtHM5pyqSAd-9SLxu-D5TCPWH/view?usp=sharing
2. Both sides of components used
https://drive.google.com/file/d/1t87USz17RDVjjuDDQqq8Swy9e6NE6O22/view?usp=sharing
https://drive.google.com/file/d/1vRBAEdZLmQEtZXzxeceMQSqt0xw1bk9q/view?usp=sharing

I thought it´s good to share this experience so ppl should more focus on HW they´re buying. Happy codding and thx for your attention.

Bodmer, 2018-10-27:
Thanks for sharing, this is helpful.  The display you have is identical to mine and is one of my favorites for the parallel interface. The ESP board has the same component placement but different manufacturer labels.

I have had bad AVR mini pro boards in the past, but they were cheap. The problem was bad or missing solder on the processor pins. Re-soldering brought them to life but that needs liquid flux to be successful.  The sockets can also be an issue especially if they are not soldered in aligned and square as this can give poor contact inside the connector strips. I re-soldered the strips on one board, it was so bad I could not insert the display shield without bending or stressing the pins!