Bodmer/TFT_eSPI #229: How can you turn display off and save power?

sst2001, 2018-10-28:
Any way to turn TFT_eSPI screen off and save power (so it can be turned on later)?

wawibu, 2018-10-28:
If the backlight is a separate pin (what it is usually) - use a pin of your controller for the backlight and turn that pin off.
Once you would like to turn the display on, just turn that backlight pin on.

sst2001, 2018-10-28:
Thx. 
Can I also save any Display activity so Display consumption goes to zero?

sst2001, 2018-10-28:
Sure, thx a lot!!!

Bodmer, 2018-10-28:
a processor digital IO pin can be used only if the display has an onboard transistor switch. If the LEDs are wired direct to the connector (often the case) they may consume 100mA or more and will overload a digital output pin. The LED will light dimly though, long term this overload may damage the processors I/O driver due to overheating of the tiny output driver transistors which are only designed to drive circa 10mA.

Please post a link to a picture of the back of your display you have, so I can see if it has a transistor fitted in the LED line. If you post a link to the vendors product page please verify the back of the display looks EXACTLY like yours, often the posted pictures on sites in China are not exactly like the display you receive, which is not helpful and may lead to the wrong advice being given.

It is quite easy to add a transistor if the display does not have one.

wawibu, 2018-10-29:
>  If the LEDs are wired direct to the connector (often the case) they may consume 100mA or more and will overload a digital output pin.

I always add a MosFET to my circuit to power the backlight - therefore I missed to tell that the backlight LED shouldn't be directly connected to a controller pin. My fault :)

Bodmer, 2018-10-29:
Yes, a logic level drive FET is needed with a suitably rated current. A FET can be used to switch off the Vcc supply to the screen as well but you must first ensure all the logic control pins connected to the display are set LOW first do they do not parasitically power the driver chip in the display. This includes the reset pin so you then cannot use the processor power on reset pin, some displays have an internal power on reset and the library uses a software reset command too as a backup for different power sequence modes. When the screen is switched back on you must the call tft.begin() to initialise it, this takes a few hundred milliseconds.

closed 2019-01-06 by Bodmer