Bodmer/TFT_eSPI #230: Touch connection

mocholor, 2018-10-30:
hope I act not to stupid here... And it's rather a question then an issue
Connected one of those cheap&cheerful red ILI9481+XPT2046 boards to Wemos D1 mini ESP8266.
Using minimal wiring, ie. only connected SCK, SDI and DC to the ESP
Works brilliantly for the display.
Now I'd like to use the touch controller. Readme says the XPT2046 shares SPI with the display and thus only Touch_CS needs to be wired  (to the pin defined in user_setup.h).
But
The XPT2046 does not seem to actually share the SPI with the display on my board, but has separate pins T_CLK, T_DIN, T_DO, T_CS. 

So do I just connect these pins to the respective SPI pins for the display
T_CLK --- Display SCK
T_DIN --- Display SDI

and an new wire for T_CS to Touch_CS
and another new one from T_DO to Display SDO/MOSI because I didn't need this until now ?

Bodmer, 2018-10-30:
Yes, you have the connections mostly right, but the T_DO line needs to go to the Wemos D1 MISO pin (where MISO = Master In Slave Out) only, do NOT connect it to the display SDO as the ILI9481 drives this line all the time.  The Wemos D1 MOSI pin will be connected to the touch T_DIN and the display SDI. Make sure you define the Touch_CS pin in the setup file.

Bodmer, 2018-10-30:
OK, hope your project is a sucess.

closed 2018-10-30 by Bodmer