Bodmer/TFT_eSPI #233: Add padding to println

abdollahpour, 2018-11-06:
Is there any way to add padding when I use _println_. When _println_ breaks a line, it jumps to column 0 of the next line. Like if I run this code, it jumps to column 0 instead of 10 for the next line:

    tft.setCursor(10, 0, 2);
    tft.println("Print something long on the screen ...");

In this way, it's almost useless to use _println_ because it breaks the design. Is there any way to add padding or create a window to print or something?

Bodmer, 2018-11-06:
There only way to do this at the moment is to print into a sprite, then plot the sprite to the screen. You then have total control of where on the screen the text goes and the same wrap rules apply inside the sprite width and height. This approach does use memory but it is possible to use a 1 bit or 8 bit sprite to reduce the amount used.

I rarely use the print stream and instead use drawString() as then you have total control of what text goes where.

Bodmer, 2018-11-06:
The other approach is to split the line in the sketch, print one line, move the cursor and print the next line etc. You then have better control of where the line is split and where each line goes on the screen.

closed 2018-11-06 by Bodmer