Bodmer/TFT_eSPI #235: ICST7735S red-tabbed 80x160 support

yknivag, 2018-11-07:
Discovered this excellent library from a tweet by iotappstory this morning and thought it would be perfect for my ICST7735S 80x160 "IPS" screen.  The screen has a red-tab on the screen protector and can be found [on AliExpress here](https://www.aliexpress.com/item/ShengYang-1pcs-IPS-0-96-inch-7P-SPI-HD-65K-Full-Color-OLED-Module-ST7735-Drive/32918394604.html).

However none of the ST7735 driver options in User_Setup.h worked - some were almost in the right place and others had the right colours, but none was perfect.

I've hacked together the attached patch and it now seems to work well with the limited testing I've done so far.

Is there anything else I should change?

[TFT_eSPI-support_red_tabbed_80x160.patch.txt](https://github.com/Bodmer/TFT_eSPI/files/2557898/TFT_eSPI-support_red_tabbed_80x160.patch.txt)

Bodmer, 2018-11-08:
Thanks, my display of that type has a green tab on the screen protector so it looks like there are a few variants of this display. I will look at your suggested changes and see if I can incorporate them.

yknivag, 2018-11-08:
Hi,

Thanks for getting back to me. I can raise a PR if that's easier for you.

I just wanted to check first that there were no extra bits of code I ought to consider.

Thanks again for a wonderful library.

Gavin.

Bodmer, 2018-11-08:
If you:
1. Update library.json and library.proprties to be version 1.1.4
2. Add the driver option to the SetupX_Template.h file
3. Generate a pull request from your fork

Then I will merge you addition into the master. Thanks.

yknivag, 2018-11-08:
I've made those changes and submitted is as PR#236.

Thank you.

Bodmer, 2018-11-13:
Merged, thanks.

closed 2018-11-13 by Bodmer