Bodmer/TFT_eSPI #238: Adding support for R61581 controller

s60sc, 2018-11-08:
One of the cheap red parallel MCU Friend 'ILI9488' boards I bought turned out to have a R61581 controller.
After looking at the MCUFRIEND_kbv library and some experimentation, the following amendments to the ILI9481_Init.h file will allow the R61581 controller to be supported by TFT_eSPI.
```
    writecommand(TFT_SLPOUT);
    delay(20);
    
#ifdef R61581_DRIVER
    writecommand(0xB0);
    writedata(0x00);
#endif
```
and
```
    writecommand(0xC0);
#ifdef R61581_DRIVER
    writedata(0x12);
#else
    writedata(0x10);
#endif
```

Bodmer, 2018-11-09:
Thanks, I will create a new driver set. I do not have one of those displays so that is helpful.

s60sc, 2018-11-10:
In addition to support rotation, with reference to the ILI9481_Rotation.h file, each TFT_MAD_SS define needs to be changed to TFT_MAD_MX

Bodmer, 2018-11-10:
Thanks, I will add that too.

Bodmer, 2018-11-25:
Added at version 1.3.5

Please test and report back if your have issues. I do not have one of these boards so I am unable to test.

Thanks for your input.

closed 2018-11-25 by Bodmer