Bodmer/TFT_eSPI #239: [question] Shades of gray ?

YordanYanakiev, 2018-11-09:
Everything is working great, but I am not sure that I completely understand the color model of the pixels, and I believed that they is using either ARGB or RGBA because uint32_t looks like it but seems like I was wrong.
Please I need to get as more as possible shades of gray. ( from completely white to completely black ) How to achieve this ?

Bodmer, 2018-11-09:
The TFTs use a 565 colour encoding [as explained here](http://www.barth-dev.de/online/rgb565-color-picker/). If you select the colour picker and set hue and saturation to 0 the you can pick shades of grey.

Red and blue are encoded as 5 bits each so you can have black and white plus 30 shades of grey. Green is encoded as 6 bits because the human eye is more sensitive to green an thus can discern smaller shades changes.

To generate shades of grey you can use the function:
uint16_t grey_color = color565( g, g, g),   // Convert 8 bit red, green and blue to 16 bit color

Where g is a number from 0 to 255.

closed 2018-11-09 by Bodmer

YordanYanakiev, 2018-11-09:
Thank You tons Bodmer !