Bodmer/TFT_eSPI #240: Supporting per project configuration with Arduino

DeanCording, 2018-11-09:
There are a couple of headaches with the current configuration process.  Firstly, it is difficult to have multiple projects that use different drivers.  Given that different drivers typically mean different screen sizes, a screen layout is usually tied to a particular screen size and driver.  Switching between projects using different screens is prone to error as you need to remember to change the settings each time.

Secondly, whenever the library is updated, all my configuration is lost.

I've looked through the previous issues and PR about this and the workaround for Platform.io.  I'd like to suggest a simple solution that works for both Platform.io and Arduino:

Add the line:

    #define USER_SETUP_LOADED 1

to the start of the User_Config.h file.  Users can then copy this file into their sketch directory and customize the configuration for their project.  

In their sketch, the user just has to have:

    #include "User_Setup.h"
    #include <TFT_eSPI.h>

to use their project's configuration. The current check in User_Config_Select.h will automatically skip including the library User_Config.h in preference for the project's User_Config.h.

per1234, 2018-11-09:
Since TFT_eSPI.cpp is a separate compilation unit from the sketch, any code that's dependent on the macros in User_Setup.h would need to be moved from the .cpp file to a header file. In the end, it might make the most sense to just make it a header-only library.

That solution has been chosen in several other popular libraries that must be configured via preprocessor directives.

DeanCording, 2018-11-09:
Turns out, that is pretty much all of the TFT_eSPI code.

It was a good idea but thwarted by the Arduino build process compiling libraries individually instead of compiling the sketch as a monolithic block.

Bodmer, 2018-11-09:
Yes, I was forced down this route by the IDE. The settings could be defined at runtime as there is plenty of FLASH in the Espressif boards but the library evolved from memory contrained AVR processors so I wanted to have better control of what sections of code get compiled. Also the setup files means the exmples do not each need to be modified for each hardware setup which removes an error prone process and aids testing.  In platformIO you van have oer project configuration as you found.
I suspect the Arduino team took the decision to compile libraries individually to avoid #define name clashes.

Thanks for thinking about this and for your efforts to help improve the library.

closed 2018-11-09 by Bodmer