Bodmer/TFT_eSPI #241: TFT 1.8" 128x160 ST7735 DrawIcon 160x128 problem

JLuisAlcaraz, 2018-11-12:
I found this problem using a DIY screen (14 pins):
When call drawIcon(iconname, 0, 0, 160, 128) it is draw with image split diagonally.
If the image to draw is smaller than 160x128 then its draw correctly.
I have TFT_RST -1

I use GREENTAB2, and found a solution changing:
ST7735_init.h
161 colstart  = 0; //2;  !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
162 rowstart = 0; //1;  !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

ST7735_Rotation.h
40  colstart  = 0; //1;  !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
41  rowstart = 0; //2;  !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Bodmer, 2018-11-12:
Thanks for reporting this. You have a display with no pixel offsets whereas the GREENTAB2 display configuration is for displays that do have offset. I can add a new GREENTAB4 option but I would like you to try the ST7735_BLACKTAB option first as that has zero offset, this will tell me if a new option is needed for your particular display.

A link to the sellers webpage will also be useful so I can see what your display and in particular the circuit board look like.

JLuisAlcaraz, 2018-11-13:
Hi,
This is the TFT I'm using:
https://es.aliexpress.com/store/product/1-8-Inch-128-160-Serial-SPI-TFT-Color-LCD-Module-Display-ST7735-With-SPI-Interface/1326062_32818686437.html?spm=a219c.search0104.3.1.226e5e5cppU5Lb&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10068_10547_319_10891_317_10548_10696_10924_453_10084_454_10083_10927_10618_10920_10921_10922_10307_10820_10301_10821_10303_537_536_5727511_10059_10884_10887_100031_321_322_10103_5727011-10891,searchweb201603_51,ppcSwitch_0&algo_expid=166e1f21-712f-4c2e-9024-8daa7f732567-0&algo_pvid=166e1f21-712f-4c2e-9024-8daa7f732567

Sometimes they sent me with a blue label.

I tested the BLACKTAB option and works perfectly!

I'm still triyng to use the smooth font option. I think I need a detailed explanation of the process. 😄

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif]<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>  Libre de virus. www.avast.com<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

Bodmer, 2018-11-13:
Thanks for the link.

For smooth fonts start with Font_Demo_1 and read the header for uploading the font files to SPIFFS. If you need help raise a new issue. Bear in mind some examples are designed for higher pixel count displays and text may wrap or run off the screen on a small display.

closed 2018-11-13 by Bodmer