Bodmer/TFT_eSPI #242: Degree symbol (°) are not displayed any moore

misosec, 2018-11-14:
Hello,
did you make any changes that could cause this issue or is it just my fault?

Bodmer, 2018-11-14:
There are no changes I can think of that would prevent a degree symbol showing.

It sounds like you have created your own vlw font. After loading the font in your sketch you can use tft.showFont(5000); and you will see the all characters are shown on the TFT with 5 seconds between pages.  This will show if the degree symbol is in the font.

If you post the vlw font file here as a zip file then I can look at it for you. Unfortunately some fonts have a different character code for the degree symbol so maybe that is the problem?

misosec, 2018-11-14:
So I tried `tft.showFont(5000)` and it shows degree symbol.
So why `String tempUnit = "°C"; tft.print(tempUnit);` doesn't work? And it worked before...but I had to reinstall Windows, so maybe I have Arduino set up incorrectly? Is it possible?

Thank you.

Bodmer, 2018-11-16:
Post you vlw font file as a compressed zip file and I will check the unicode code point value, usually it is 0x00B0. So "°C" should work. If showFont can print it on the screen then you will be able to as well, it is a case of using the right point code.

Bodmer, 2018-11-16:
Make sure you have not included the Masculine Ordinal Indicator "º" by mistake, this looks like a degree "°" symbol!

misosec, 2018-11-16:
Hello,
so here is that font:
[accid30.zip](https://github.com/Bodmer/TFT_eSPI/files/2588342/accid30.zip)

Could you check it out for me please?
Thank you.

misosec, 2018-11-16:
Hello again,
so this works: `String tempUnit = " \xB0""C";`
Great!

Thank you.

closed 2018-11-16 by misosec

Bodmer, 2018-11-16:
Thanks for posting the solution. The Arduino IDE has had a bug where it does not compile in the right code point if the sketch is not saved before compiling. So maybe that is one reason for the missing symbol. I am not sure if that IDE bug is still present but it is something that is worth trying.

maltegwde, 2018-12-09:
I'm having a similar problem. All charactes are shown correctly when calling `tft.showFont(2000);` But when I'm trying to display them only yen + dollar are shown correctly.

```
String pound = "\xc2a3", yen = "\xc2a5", dollar = "\x24", euro = "\xe282ac";
tft.print(pound);
tft.print(yen);
tft.print(dollar);
tft.print(euro);
```

Any idea what I'm doing wrong?

Bodmer, 2018-12-09:
The unicode for euro is 0x20AC, tyically you can use you keyboard to enter characters like this bo using the Alt key. The pound sign is 0xA3.

If you are using create font the the [character code ranges](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/Create_Smooth_Font/Create_font/Create_font.pde#L305) for the font you are creating need to be correct as well.

Bodmer, 2018-12-10:
I am not able to try this at the moment but I suspect the answer is to use the UTF8 coded byte sequence like this:
"\xe2\x82\xac"

If you can type the Euro charecter directly using your keyboard with an [Alt sequence](https://usefulshortcuts.com/alt-codes/currency-symbol-alt-codes.php) then that should work, or cut and paste this into your code.
"€"

maltegwde, 2018-12-12:
Neither of them works for me, I'm just going to use "Euro" now.
Thank you for taking the time and trying to solve these problems!

muellest, 2019-04-15:
Is there a solution for this?

Bodmer, 2019-04-15:
If the font contains the characters then you can use the string or the 16 bit code in a string, for example:
```
  tft.print("°");        // °
  tft.print("\u00B0");   // °

  tft.print("£");        // £
  tft.print("\u00A3");   // £

  tft.println("€");      // €
  tft.println("\u20AC"); // €
```

pipi61, 2019-04-15:
Try this:
use http://fontforge.github.io/en-US/downloads/  to edit ttf fontfile and move character to another (printable) position, Generate new ttf fontfile.
Use from Adafruit-GFX-Library  fontconvert utility to generate new C source from new ttf file.
I use this method to generate the new printable international charset (move international characters to less than 0xFF)

erickzind, 2019-09-05:
> Try this:
> use http://fontforge.github.io/en-US/downloads/ to edit ttf fontfile and move character to another (printable) position, Generate new ttf fontfile.
> Use from Adafruit-GFX-Library [...]

There are no changes I can think of that would prevent a [degree symbol](https://www.degree-symbol.com/) showing.

Thanks @pipi61  I solved my problem.