Bodmer/TFT_eSPI #244: Please Support for ST7789V (240x320) TFT Driver too

TridentTD, 2018-11-16:
When I use TFT-ST7789 driver (240x240) for my TFT-ST7789V(240x320) ,
it occurs inverted colors.

Please support ST7789V (240x320) driver too.
Thank you.

Bodmer, 2018-11-16:
The 240x240 and 240x320 displays use a different colour order. The[ size of screen must be specified](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L32) for the library to change the colour order appropriately.

Bodmer, 2018-11-19:
I have [added a new option to version 1.2.2 to define the colour order](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L32) for the ST7789 as there seems to be a lot of manufacturer variants.

See also issue #246 for the tft.invertDisplay(1); solution if white and black are swapped.

closed 2018-11-19 by Bodmer

ed-french, 2020-04-14:
Possibly not a helpful observation, but I had hells delight today with sn ST7789V display on a TTGO T-Display board with colour handling. Specifically, when I cleared the screen to bright red, it would often come out as bright blue. Changing a number that was printed to the screen in the loop changed what colour the screen would be. Adding the tft.invertDisplay(1); fixed the problem.
If I get the chance I will try and write a proper test case, but in the following loop:

`    
    uint32_t col=0x1; 

    while (true)
    {
        col=0x1;
        for (uint32_t i=0;i<16;i++)
        {
            tft.fillScreen((uint32_t)col);
            tft.setTextColor(DISPLAY_WHITE,(uint16_t)col);
            tft.setCursor(50,50);
            tft.print(col);
            delay(300);
            col=col*2;
        }
    }
`

Changing from tft.print(col) to tft.print(i) changed the colours of the fillScreen!
So far, adding the invert statement has made this consistent.