Bodmer/TFT_eSPI #245: Workaround to make ILI9341 parallel with TP28017 (analog resistive) work

ddrmax, 2018-11-16:
Hello @Bodmer ,
During some time in my work, I made a 2.8" 320x240 ILI9341 parallel display with a TP28017 TouchScreen (a one that uses a reading of analog values) work on a ESP32 without any screen refreshing crash.
link to the [screen:](https://www.amazon.fr/ELEGOO-Tactile-320x240-Technical-Arduino/dp/B01JD4TJZU/ref=pd_ybh_a_12?_encoding=UTF8&psc=1&refRID=DPCC4344S00GZ9R8NPTK)
This could be a good way to resolve the issue #106.

The mods that I have done are: 

- modify the adafruit Touchscreen Library  to correlate the adc bandwith (12 bits instead of 10).

- Add a touchscreen maping to the screen size.

- added a workaround for the pin state that check the state of the touchscreen pins and restore them after calling the TouchScreen::getpoint() function.

I will be doing a fork of your repo and do a push request when my workaroud is no more a dirty hack.

Bodmer, 2018-11-16:
Thanks, I do not have one of those screens so will be unable to test. Can you fork the Adafruit library too and make the needed changes. An example sketch for TFT_eSPI will also be needed that includes a link to your Adafruit touch screen variant.

ddrmax, 2018-11-16:
Ok when I'm done, i will make the push request with an user setup and a I surely will add a new example as well (for example the TFT_Clock with buttons to change hours / mins / sec

Normally I will make another change: your Touch.h and . cpp will become warpers for your actual touch library and the modified adafruit that if no touch defs are used the they will be Dummy functions and for example i will add two new defs:  TOUCH_SPI and TOUCH_ANALOG

- if TOUCH_CS is defined it will automatically define TOUCH_SPI 

- if one of the analog pin defines are defined like TOUCH_XP or TOUCH_YM it will automatically define TOUCH_ANALOG

Another Thing is that I will make a push request as well to the Adafruit Touchscreen repo since an issue is open for ESP32 support.

Bodmer, 2018-11-20:
OK, I can see you have thought this through very well. All sounds good.

Bodmer, 2018-11-24:
Post back when you have made progress.

closed 2018-11-24 by Bodmer

cipibad, 2018-12-04:
@ddrmax, could you share your code?
I am new to ESP32 development and have troubles with screen refresh after reading touchscreen on similar setup.

Thanks

ddrmax, 2018-12-05:
I have put my initial work here https://github.com/ddrmax/teste_spi

This is quite a dirty code since for now (until I get my hollidays in the begining of january) I can't prepare a more cleaned code to do a push request.

to get at least the screen to refresh (in a safer way) when you to get a point use this function instead of calling directly the adafruit one :
```
//add this before setup() and loop();
int D1State,D0State,DCState,CSState;
TSPoint Getpoint()
{
    D1State=digitalRead(YM);
    D0State=digitalRead(XP);
    DCState=digitalRead(XM);
    CSState=digitalRead(YP);
        TSPoint p = ts.getPoint();
pinMode(XP, OUTPUT);
  pinMode(XM, OUTPUT);
  pinMode(YP, OUTPUT);
  pinMode(YM, OUTPUT);
    digitalWrite(YP, CSState);
  digitalWrite(XM, DCState);
    digitalWrite(XP, D0State);
  digitalWrite(YM, D1State);
return p;
}

```
keep in mind that you have to map the returned x and y values to the size of the screen like i have done:
```
//A large part comes from a modified example provided in my elegoo CD that came with the screen
//Touch For New ILI9341 TP
#define TS_MINX 120
#define TS_MAXX 900

#define TS_MINY 70
#define TS_MAXY 920

//maping used after getting the touchscreen info
bool pressed = (p.z>10)?true:false;
  int16_t t_x = map(p.x, TS_MINX, TS_MAXX, tft.width(), 0);
  int16_t t_y = (tft.height()-map(p.y, TS_MINY, TS_MAXY, tft.height(), 0));
```

cipibad, 2018-12-05:
Thank you for help.
I'll dig into it, and provide link to working example, if/when I'll manage it:)

LKAT1, 2023-11-18:
hey I was trying to create a UI for the same/similar touchscreen, I was having an issue where after one button is pressed the way the coordinates are read is flipped, reading touch inputs from the bottom right instead of the top left
`// You need to just currentscreen vairaible,
//if statements

// create numbers for the vari
#include <Adafruit_GFX.h>
#include <Adafruit_TFTLCD.h>
#include <TouchScreen.h>
#include <stdint.h>

#define LCD_CS A3
#define LCD_CD A2
#define LCD_WR A1
#define LCD_RD A0
#define LCD_RESET A4
// these pins define the pins being used by the Arduino^^
#define TS_MINX 118
#define TS_MINY 106
#define TS_MAXX 950
#define TS_MAXY 933
#define YP A3
#define XM A2
#define YM 9
#define XP 8
// this code calibrates the screen
Adafruit_TFTLCD tft(LCD_CS, LCD_CD, LCD_WR, LCD_RD, LCD_RESET);

TouchScreen ts = TouchScreen(XP, YP, XM, YM, 300);
// these define the pins used to activate the screen
#define BLACK 0x0000
#define BLUE 0x001F
#define RED 0xF800
#define GREEN 0x07E0
#define CYAN 0x07FF
#define MAGENTA 0xf81F
#define YELLOW 0xFFE0
#define WHITE 0xFFFF

int currentscreen = 0;
bool Camera = 0;
bool tracker = false;
bool Homevar = false;
bool backbutton = 1;
bool welcome = true;

void Home() {
  tft.fillScreen(RED);
  tft.setCursor(60, 100);
  tft.setTextSize(2);
  tft.print("CHOOSE APPLICATION");
  tft.fillRect(10, 10, 75, 75, BLUE);
  tft.fillRect(125, 10, 75, 75, BLUE);
  tft.fillRect(240, 10, 75, 75, BLUE);
  // creating buttons for applications, first test out using LED's
}
void Welcome() {
  tft.fillScreen(WHITE);                //WHat to fill the screen colour with- colours stated above
  tft.drawRect(0, 0, 319, 240, WHITE);  //the rectangularshape that the screen fills
  tft.setCursor(5, 5);                  //set cursor is where text will begin on the screen, top right corner of the text
  tft.setTextColor(BLACK);
  tft.setTextSize(2);
  tft.print("WELCOME LKAT");
  //add parameters
  tft.fillRect(50, 180, 210, 40, BLACK);
  tft.drawRect(50, 180, 210, 40, GREEN);
  tft.setCursor(60, 190);
  tft.setTextColor(WHITE);
  tft.setTextSize(2);
}

void LED() {
  tft.setCursor(5, 5);
  tft.fillScreen(BLUE);
  tft.print("HELLO");
}

void setup() {
  Serial.begin(9600);
  tft.reset();
  uint16_t identifier = tft.readID();
  tft.begin(identifier);
  tft.setRotation(1);  //changing this value between 1 and 0 will change the layout of the text on screen. landscape portarait
  currentscreen = 0;

  Welcome();
}

void loop() {

  TSPoint p = ts.getPoint();
  if (p.z > ts.pressureThreshhold) {
    p.y = map(p.y, TS_MINY, TS_MAXY, 0, 480);
    p.x = map(p.x, TS_MINX, TS_MAXX, 0, 320);
    pinMode(XM, OUTPUT);
    pinMode(YP, OUTPUT);

    Serial.println(p.x);
    Serial.println(p.y);
    //set screen resolution

    if (currentscreen == 0) {
      if (p.x > 50 && p.x < 260 && p.y > 180 && p.y < 260) {

        currentscreen = 1;

        Homevar = true;
        Home();
      }
    }

    if (currentscreen == 1 && Homevar == true) {

      //check coordinates, screen has rotated
      if (p.y > 10 && p.y < 85 && p.x > 10 && p.x < 85) {

        LED();
        currentscreen = 2;
      }
      //other if
    }
  }
}`