Bodmer/TFT_eSPI #246: Inverted colours ST7789 240x240

alexferreira1, 2018-11-18:


alexferreira1, 2018-11-18:
Can display image but colours are inverted. I have gone into the user setup file and change the height and width to 240 but no luck.

alexferreira1, 2018-11-18:
https://www.aliexpress.com/item/1pcs-1-3-inch-7PIN-SPI-HD-Full-Color-IPS-TFT-Display-Screen-ST7789-Drive-IC/32861463404.html

Bodmer, 2018-11-18:
Hi, thanks for reporting this. It appears there are many variants of this display that have different color orders so I am going to have to add a new setup parameter. Until I can do this you can correct the colour by [editting this line 43 in the library file](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7789_Defines.h#L43) to read:

#define TFT_MAD_BGR 0x00

You must then include setRotation in your sketch setup() to force the colour order to be correct.

alexferreira1, 2018-11-18:
I have added "tft.invertDisplay(1);" just after "tft.init" which seems to have corrected it. I can use this as a temporary fix until I try editing the library.

Bodmer, 2018-11-18:
Ok, I see now that the problem is not the RGB colour order but a complete inversion of colours. So black becomes white etc. 

Thanks for reporting the solution.

closed 2018-11-19 by Bodmer

marcosgadze, 2022-12-20:
I had the same problem and i changed the setup file, in my case the one that is in the folder "TFT_eSPI_Setups" and the name "Setup43_ST7735_ESP32_128x160" but i think this works for all the other setup files.
I enabled the line "#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue" that was with //. Also you can try the second line BGR.
I tried first to change the "User_Setup.h" file in the folder "TFT_eSPI-master" but it didn't work, because for the setup of the program, it commands the TFT_eSPI_Setups folder.