Bodmer/TFT_eSPI #247: Wemos D1 Mini and ILI9341

SWEUSPeter, 2018-11-18:
Hi,
I came across your excellent implementation of DarkSky and decided to update my old weather station based on ThingPulse. However, I'm missing something simple since I cannot get the display to work. I change the settings to ...
// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
#define TFT_CS   PIN_D1  // Chip select control pin D8
#define TFT_DC   PIN_D2  // Data Command control pin
//#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

I came to this conclusion after my current setup is ...

Wemos D1 Mini | ILI9341
-- | --
D4 | T_IRQ
D6 | T_DO
D7 | T_DIN
D3 | T_CS
D5 | T_CLK
D6 | SDO
D8 | LED
D5 | SCK
D7 | SDI
D2 | D/C
RST | RESET
D1 | CS
GND | GND
3V3 | VCC

Please let me know what I'm doing wrong :-)

Peter

Bodmer, 2018-11-18:
You probably need to set D8 high to turn on the LED backlight and also D3 high to disable the touch controller. The other pins look OK to me. Report back if that does not work.

ThingPulse have created a library that uses Open Weather Map but I have not tried it.

Post

SWEUSPeter, 2018-11-18:
That fixed it! Thank you for a quick reply.
As for Open Weather I didn't really like what I saw with the options available. Came across DarkSky and liked their API and the options available.
Looking forward to some of your either ideas to expand the design. In the meantime I will make my own changes.

Peter

Bodmer, 2018-11-19:
Great. Suggestions for improvements are always welcome.^

closed 2018-11-19 by Bodmer