Bodmer/TFT_eSPI #248: Why does function  tft.readPixel() alway return 0?

TridentTD, 2018-11-19:
When I read color of the pixel by function tft.readPixel() ,
it always returns 0 value.

How to solve this?
Thank you.

ddrmax, 2018-11-19:
Hello, the tft.readpixel( x,y) only works with certain drivers.
for example in ESP32 in  parallel mode only these drivers works:

- ILI9341

- ILI9488

- ILI9481 

in SPI mode, the same drivers seems supported (same type of read but with and ILI9488 workaround)

if the screen you use is the same as the other issues you posted (ST7789/V) i think it's not supported nor it has to be implemented

TridentTD, 2018-11-19:
ddrmax .
Thank you .

Bodmer, 2018-11-19:
As @ddrmax has pointed out only certain screens are supported for reads. It is difficult to predict which screens will read OK as it depends in some cases how the PCB has wired up the controllers mode pins.

Displays that have a bidirectional SDA pin cannot be read by the library. This situation may changes if I can figure out how to setup both the ESP8266 and ESP32 to handle a single bidirectional SPI data pin.

You question has highlighted the fact that the read capability and limitations are not covered in the ReadMe, I will fix this soon.

TridentTD, 2018-11-20:
How can I check my display have a bidirectional SDA pin ?

If my display(ST7789V 240x320) 's pin set like the following .

TFT_MISO  19  // Master Input Slave Output pin
TFT_MOSI  23  // Master Output Slave Input pin
TFT_SCLK  20  // Signal Clock pin
TFT_CS    5      // Chip select control pin
TFT_DC    16   // Data Command control pin
TFT_RST   17   // Reset pin (could connect to RST pin)
TFT_BL    15    //  BlackLight pin

I wish to get this display's screenshot.
Can this display support  function tft.readPixel() ?

Thank you @Bodmer 

(BTW. Sorry for my poor English.)

Bodmer, 2018-11-20:
A board with an ST7789 on arrived yesterday in the post so I will investigate the options on that screen in the next few days. It appears your board does not have a bidirectional data bus but separate MOSI and MISO pins so I expect I can support screen reads.

One user with an ESP32 with extra PSRAM fitted created a full screen buffer in RAM, drew the graphics into it and read back the resulting image. But this can only be done with extra PSRAM as the ESP32 does not have sufficient RAM allocated to sketches without it.

Bodmer, 2018-11-21:
The board I have has a named SDA pin only which is bidirectional. What display do you have (please provide a link to the vendors page) and what are all the pin names on the interface?

TridentTD, 2018-11-21:
https://www.aliexpress.com/item/2-4-inch-10P-262K-SPI-TFT-LCD-Screen-ST7789-Drive-IC-240-RGB-320/32824463892.html?spm=2114.search0104.3.311.115521ddK2028j&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_319_317_5733915_10696_10924_10084_453_454_10083_10618_5733715_10920_10921_10307_10922_537_536_5733815_10059_10884_5734015_10887_100031_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=da68ea69-c07c-4d39-8685-390184c28732-44&algo_pvid=da68ea69-c07c-4d39-8685-390184c28732

Bodmer, 2018-11-21:
OK, that link is very helpful since it lists the pins for the display. The display you have has a bidirectional SDA  pin which will be connected to the ESP32 MOSI pin. The ESP32 MISO line is not connected to the display and that is the pin that is normall used to read from an SPI device. 

My research indicates the ESP32 does not have a bidirection SPI pin option but I have managed to read my display by reading bit by bit using software. I will tidy up the code within the next 24 hours and upload for you to try. I will post a message here when it is done.

Bodmer, 2018-11-22:
I have a version of the library working but there is some rare pixel colour corruption that suggests a read timing violation. I will therefore need more time to track down the poblem!

Bodmer, 2018-11-22:
Interestingly I have readRect() working fine with the ESP32 and ESP8266 but even though the same read function is shared with readRectRGB() that latter function appears to be getting one corrupted pixel at the start of every block request on the ESP32 only.  This one is a nice puzzle for me to solve! I will look again later today and create a diagnostic sketch.  I also seem to have uncovered an anomaly in the ESP8266 core libraries that makes reading pixels 10x slower than it should!

I may create a branch for you to try the working function if I cannot make progress. The read performance with the ESP32 is quite good for a software solution.

TridentTD, 2018-11-23:
Yes, very interting  in your work!
I use this TFT with ESP32, do you mind to create branch for trying ?

TridentTD, 2018-11-23:
Thank you very much @Bodmer 

I try readPixel() , it can read color.
However I have some question , I test by the following code.

```cpp
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); 
void setup() {
  Serial.begin(115200);
  tft.init();
  tft.fillScreen(TFT_BLACK);
  tft.fillRect(10,10,50,50, TFT_BLUE);       //  <----- Test color
  Serial.println(tft.readPixel(30,30),HEX);  // <----- Try to read color
}

void loop() {
}
```

When I test by TFT_BLUE ( 0x001F ) 
but when read color shows  0x1E.

and other test color...
TFT_BLUE ( = 0x001F )     -------> color read ---> 0x1E
TFT_GREEN ( = 0x07E0 )    -----> color read ----> 0x0FC1
TFT_RED (=0xF800 )   ---------> color read ----> 0xF000

The read-colors is near color with the test-color.
Why these read-color are not match to the test-color?

Thank you.

Bodmer, 2018-11-23:
I will run the same test.  It looks like there is a single bit shift in the colour value. This could be a bug or because the ST7789 TFT (like some others) needs an extra clock pulse during the read, so the first byte is read with 9 clock pulses discarding the MSB and the rest with 8.

Unfortunately the ST7789V data sheet I found on the internet does not seem to have much information on SPI reads, it talks about parallel reads and so I see nothing about needing an extra clock pulse.  I will have to dig further, see if I can find a better datasheet and run some tests but your results certainly suggest this extra clock pulse is needed.

Bodmer, 2018-11-24:
I have updated the library and came up with a neater and faster solution for the ESP32.

I have run this block move test and it runs fine:
[Block_move_sketch.zip](https://github.com/Bodmer/TFT_eSPI/files/2611943/Block_move_sketch.zip)

The settings I used are:

```

#define ST7789_DRIVER

#define TFT_SDA_READ // Read from display that only provides an SDA pin

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18

#define TFT_CS   05
#define TFT_DC   16
#define TFT_RST  17

#define TFT_WIDTH  240
#define TFT_HEIGHT 320

//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed for the ST7789V
```

Let me know if it works for you.  At the moment only the ESP32 has been tested.

closed 2018-11-24 by Bodmer

TridentTD, 2018-11-24:
Example Block_move_sketch.zip  
it runs fine now. Thank you.

If I want to save screenshot to SD-Card  ( jpg or bmp format),
when capture screen by readRect()  , how to save these data to jpg or bmp file ?
Thank you .

Bodmer, 2018-11-24:
Great.

A new library would be needed to save images in a standard format. I have no plans at the moment to do that.  BMP format would be the simplest to implement in an uncompressed format.